adafruit/circuitpython #570: Add time.gmtime

tannewt, 2018-02-01:
Our time module is missing gmtime which converts a number of seconds after January 1, 1970 to a `time.struct_time`: https://docs.python.org/3/library/time.html#time.gmtime

The Python-> conversion would be added here: https://github.com/adafruit/circuitpython/blob/master/shared-bindings/time/__init__.c

and the implementation in a new `time/__init__.c` here: https://github.com/adafruit/circuitpython/tree/master/shared-module

It'd make sense to support converting to and from struct_time because our RTC libraries use it.

brentru, 2018-02-02:
I'll give this a shot.

fogmansac, 2018-02-04:
I am curious how you envision this working.  As I understand it - and I may not:
1. gmtime takes as its parameter the number of seconds since the epoch and returns a time.struct_time eith the current date and time.
2. The epoch is commonly midnight Jan 01, 1970.  
3. As near as I can tell CircuitPython has no epoch.  
4. Circuit Python also has no time.time().  Though on the Metro, time.monotonic() returns what appears to be the  number of seconds since the last power up of the board.  
Were you envisioning a somehow implementing time.time() using monotonic and some initializer to create a sort of pseudo epoch or just a way to turn monotonic to days, hours, minutes etc.?  The root of what I am struggling with is how without some source of real time to get the number  of seconds since the epoch this is useful.  If you have an RTC - DS3231 etc - why would you want this since you get a time structure with it all there?

John

tannewt, 2018-02-04:
@fogmansac Its meant as a simple conversion from standard Jan 01, 1970 epoch designation.

It came up in Discord chat because someone was fetching the time from a website that returned it as seconds from epoch.

brentru, 2018-02-07:
I've been working on this today, but @dhalbert  noticed that using the standard epoch designation (1970) won't work if we pass it anything larger than 2^30 (since the top bit is a sign bit).

We think it _has to either use a later epoch than 1970_, or maybe shifting the epoch (https://howardhinnant.github.io/date_algorithms.html#civil_from_days) is a way to avoid this as well.

tannewt, 2018-02-07:
I don't think we should use a different epoch. Lets comment it with a warning about int size and make sure we handle the larger ints available on m4 boards.

dhalbert, 2018-02-07:
But we can't use the 1/1/1970 epoch to convert the current date/time, because the number of seconds to now since 1/1/1970 is > 2**30. We can only do dates up to sometime in 2014. If we think we're going to get values from some other source that are >2**30, we have to think about what to do.

tannewt, 2018-02-08:
On the M4 boards we can store ints > 2**30. We could provide a separate method that could handle the conversion from a series of ints or bytes if we want to handle the M0. time.gmtime should match what CPython does regardless of whether the input ints can be current time or not.

tuupola, 2018-02-11:
32bit unsiged integer could store timestamps until year 2106. With unsigned you lose the possibility to represent days before epoch though.

tannewt, 2018-02-13:
@tuupola CircuitPython's ints are 31 bit signed so we only get 2**30 without long int support enabled.

tuupola, 2018-02-13:
Oh, did not know. Just out of interest where did the one bit go?

tannewt, 2018-02-13:
We lose bits due to pointer boxing where data is stored in pointers themselves. There is a bit of explanation here: https://github.com/adafruit/circuitpython/blob/master/py/mpconfig.h#L66

gpshead, 2018-05-15:
We could deal with the 1970 epoch on all platforms if we don't mind losing accuracy as time goes on by allowing floats.  This is similar to what `time.monotonic()` is currently doing by returning a float uptime.  The larger the value, the less precise it'll be. :/  Currently that appears to put the resolution at a little over two minutes.

```python
>>> now = 1526400902.0523624
>>> now
1.5264e+09
>>> '%0.9f' % now
'1526401042.938232421'
```
(that was on my M4 with 3.0.0-a6)

Whenever an API matching a CPython stdlib name is used, it should really behave as close as possible to the expectations of that. A function using a more modern epoch (such as 2018) might make sense for CircuitPython or MicroPython but should use a new name.

notro, 2018-05-17:
I agree it's better to disable the functionality than to return a value that's 2 minutes off.

Is there a plan to enable long int?
```
Adafruit CircuitPython patchbase-dirty on 2018-05-10; Adafruit Feather M0 Express with samd21g18
>>> now = 1526400902
OverflowError: long int not supported in this build
>>>
```

tannewt, 2018-05-17:
@dhalbert You've tried enabling long ints on express builds right? Do you think we have enough space?

dhalbert, 2018-05-17:
Longints are currently enabled on 3.0 Express builds. I checked again recently about enabling on non-Express builds: https://github.com/adafruit/circuitpython/issues/572#issuecomment-387505388, quoted below. It looks pretty iffy that it would ever fit on 192KB builds (non-Express).

>Longints are not going to fit in the non-Express builds for 3.0 without removing other stuff. I recently tried MPZ and LONGLONG. Currently a Gemma M0 build has about 1800 bytes free. If I enable MPZ, it overflows by 2900 bytes. If I turn on -finline-limit, I can get it lower, but it still overflows by about 1k. And this is without touchio yet. (LONGLONG overflows by more.)

dhalbert, 2018-05-17:
@notro, oops, you're right, it's enabled on M4, but not M0 Express.

notro, 2018-05-17:
> Can someone please change it to 1970 and remove the APIs in non-express builds?

I can do this. Is there a 'long int in use' macro I can use to #ifndef the functionality?

notro, 2018-05-17:
Maybe this?

```diff
 STATIC const mp_rom_map_elem_t time_module_globals_table[] = {
     { MP_ROM_QSTR(MP_QSTR___name__), MP_ROM_QSTR(MP_QSTR_time) },
 
     { MP_ROM_QSTR(MP_QSTR_monotonic), MP_ROM_PTR(&time_monotonic_obj) },
     { MP_ROM_QSTR(MP_QSTR_sleep), MP_ROM_PTR(&time_sleep_obj) },
     #if MICROPY_PY_COLLECTIONS
     { MP_ROM_QSTR(MP_QSTR_struct_time), MP_ROM_PTR(&struct_time_type_obj) },
+    #if MICROPY_LONGINT_IMPL != MICROPY_LONGINT_IMPL_NONE
     { MP_ROM_QSTR(MP_QSTR_localtime), MP_ROM_PTR(&time_localtime_obj) },
     { MP_ROM_QSTR(MP_QSTR_mktime), MP_ROM_PTR(&time_mktime_obj) },
+    #endif
     #endif
+    #if MICROPY_LONGINT_IMPL != MICROPY_LONGINT_IMPL_NONE
     { MP_ROM_QSTR(MP_QSTR_time), MP_ROM_PTR(&time_time_obj) },
+    #endif
 };
```

dhalbert, 2018-05-17:
@notro that looks good!

tannewt, 2018-07-03:
This is done. Thanks all!

closed 2018-07-03 by tannewt