adafruit/circuitpython #574: Update Intro guide to include PyCharm MicroPython REPL

kattni, 2018-02-02:
Add a page to the Intro guide explaining how to get the REPL working in PyCharm with the MicroPython plugin.

kattni, 2018-02-02:
Guide page is published. However, screenshots are still being updated. Leaving issue open until guide is entirely complete.

kattni, 2018-02-03:
Screenshots have been updated, guide page here: https://learn.adafruit.com/welcome-to-circuitpython/pycharm-and-circuitpython. Blog post about update has been published here: https://blog.adafruit.com/2018/02/02/guide-update-get-connected-to-the-repl-using-pycharm/

All set to go!

closed 2018-02-03 by kattni

pmcanseco, 2019-03-02:
Hey there, the learn.adafruit.com link is broken, but the blog post is still up. What's the corrent learn.adafruit.com link?

siddacious, 2019-03-02:
@pmcanseco both of the links above work for me as does the one in this blog post:
https://blog.adafruit.com/2018/02/02/guide-update-get-connected-to-the-repl-using-pycharm/

Are you talking about a different link? If so please paste it here.

ladyada, 2019-03-02:
@kattni made it private because it stopped working (i think?)

kattni, 2019-03-02:
I removed the page because it never really worked properly. It was inconsistent and for me, at least, required deleting all of my settings to get it to work each time. It was never designed to work with CircuitPython, it was designed to work with MicroPython, so it provides a lot of features that are irrelevant to CP such as flashing the board.

If you wish to access the REPL from PyCharm, consider using the built in terminal and `screen` to connect to the serial console/REPL.