adafruit/circuitpython #576: remove RTD 'stable' version

ladyada, 2018-02-03:
there's some old rev documentation in RTD ? not sure whats up but

https://circuitpython.readthedocs.io/en/stable/shared-bindings/touchio/TouchIn.html

is not the same as

http://circuitpython.readthedocs.io/en/2.x/shared-bindings/touchio/TouchIn.html

e.g. **raw_value** and **threshold** are missing from 'stable'

here's what it indexes

https://readthedocs.org/projects/circuitpython/

tannewt, 2018-02-04:
Stable is now deactivated but the link still works. The version bar is red. Want that for 1.x as well?

ladyada, 2018-02-04:
yes please make 1.x red

willingc, 2018-02-04:
@tannewt You should be able to remove stable from displaying at all through the RTD UI. You should be able to activate/deactivate each build as needed. If you send me screenshots of the UI settings, I can see the config.

You can add warnings on the top of files by using [`rst_prolog`](http://www.sphinx-doc.org/en/stable/config.html#confval-rst_prolog). You may need to write a bit of code in conf.py to select a subset of source docs if desired. [IPython's `conf.py`](https://github.com/ipython/ipython/blob/master/docs/source/conf.py) has examples of its use.

tannewt, 2018-02-06:
Found it! We "cleaned" the stable copy and deactivated 1.x. Thanks for the tips @willingc!

ladyada, 2018-02-06:
thanks everyone :)

closed 2018-02-06 by ladyada