adafruit/circuitpython #577: Consider adding a `storage.format()` to erase and create a fresh filesystem

dhalbert, 2018-02-03:
It would be nice to be able to erase and re-create a damaged filesystem from the REPL or a program. Currently on non-Express boards we need to erase all of flash and then reload CircuitPython. On Express boards we have a special Arduino program that erases SPI flash, and then we need to reload CPy.

Something like `storage.format(<filesystem>)` would be helpful. One minor issue is how to specify the filesystem if it's so damaged we don't know what it's named.

Some previous discussion: https://github.com/adafruit/circuitpython/issues/386#issuecomment-356731896 ff.

tannewt, 2018-02-05:
> Some previous discussion: #386 (comment)

Could we specify it by mount point? That's consistent with remount I
believe.

dhalbert, 2018-02-05:
The question I had was whether we can identify the mount point name if it's not mountable due to corruption. But I haven't looked at the code yet.

tannewt, 2018-02-05:
Ah good point! We could hard code it but that'd be a bit weird.

Have any other ideas? We could fake /dev/disk/* but thats probably overkill.

PR #747 (2018-04-10): add storage.erase_filesystem() to erase and reformat CIRCUITPY
Fixes #577.

Note that `storage.erase_filesystem()` does a `microcontroller.reset()` after reformatting. This is necessary to get the host computer synced up with the filesystem, including both contents and mountpoint name.

The main use of this is to be called from the REPL, since if the filesystem is damaged, it's not likely one can write a file to call this anyway.

I purposely chose a long name to make it less likely to be used casually.

#577 discusses adding `storage.format(...)`, which could also be used on the SD card. That [... cut]
closed 2018-04-10 by tannewt