adafruit/circuitpython #578: Library Request: Matrix Keypad

sommersoft, 2018-02-04:
A member on Discord was asking about using a Matrix Keypad with CircuitPython, similar to the [Adafruit Membrane 3x4 Matrix Keypad](https://www.adafruit.com/product/419) (PID: 419).

* The Adafruit product page references this Arduino library to use with it: [Keypad Library for Arduino](http://playground.arduino.cc/Code/Keypad)
* The GitHub repo for the Arduino library is here: [Keypad](https://github.com/Chris--A/Keypad)
    * It is licensed under GNU GPL, so be sure to include that license information.

deshipu, 2018-02-04:
I wonder if we could extend the `gamepad` library to support matrices.

arturo182, 2018-02-06:
I wonder if it won't be a little confusing to have to use a library named `gamepad` for key matrices, maybe it would be better to keep the `Gamepad` class but rename the library and add the `Keypad` class in that library. Both classes are pretty close in functionality so they would reuse a lot of the code, like the key scanning and the timer tick stuff.

ladyada, 2018-05-24:
hey i need to make one anyways, but it'd be in pure python - so you have to scan it whenever you wanna use

ladyada, 2018-05-29:
https://github.com/adafruit/Adafruit_CircuitPython_MatrixKeypad

closed 2018-05-29 by ladyada