adafruit/circuitpython #579: CircuitPython and Circuit Playground Express Ikea lamp project

kattni, 2018-02-04:
This idea was pulled from a Twitter post about someone using an old Ikea lamp and a CPX to create a colorful lamp. That lamp was no longer available, so we've chosen 3 more to work with to see what we can do with them.

I have obtained the lamps, and am working on seeing how the CPX fits into them and what different ways I can interact with them. One is meant to try to use sound, light, motion and touch if possible to to different things. The other two are meant to use with IR.

kattni, 2018-02-04:
The smaller lamp consists of a silicone mold (shaped like a little creature!), and a hollow hard plastic bit that fits inside containing the electronics. When the plastic bit is (easily) removed, the Circuit Playground Express fits exactly into a groove along the bottom of the silicone mold, and the USB cable fits through the charging port. Thanks, Ikea!

After experimentation, the first lamp is limited to motion as an input: tap, shake, and acceleration. I'll be documenting in the guide the various other inputs attempted and why they failed.

I've used generators to create a series of animations and solid colors using tap and shake to cycle through the colors/animations and different brightness levels respectively. 

I'm now working to create a state machine for use with rotation states as an input - the idea being if you rotate it 90 degrees for a set amount of time, it sends an input, then you must return it to it's upright state before you can utilise the 90 degree input state again. This will be added to the current code to create the final program for this lamp.

ladyada, 2018-02-05:
hiya @kattni put the updates in the basecamp thread. thanks!

kattni, 2018-02-22:
Finished up! https://learn.adafruit.com/hacking-ikea-lamps-with-circuit-playground-express/

closed 2018-02-22 by kattni