adafruit/circuitpython #581: add  driver for STMPE610

jerryneedell, 2018-02-05:
Add a driver for the resistive touch screen stmpe610.  You can assign this to me.

ladyada, 2018-02-05:
hi! yep this is a good idea :) i did 'FT62xx' called "FocalTouch" 
there's *two* interfaces i2c and spi but we tend to use SPI
and SPI is *wierd* on the STMPE is the 'polarity' depends on startup conditions. so you might end up in MODE1
https://github.com/adafruit/Adafruit_STMPE610/blob/master/Adafruit_STMPE610.cpp#L104

ladyada, 2018-02-06:
for the return values, i think a dict with 'x' 'y' and 'pressure' might best
https://github.com/adafruit/Adafruit_CircuitPython_FocalTouch/blob/master/adafruit_focaltouch.py#L96
that way its somewhat compatible with focaltouch?

jerryneedell, 2018-02-06:
Thanks for the pointers. I’m looking forward to getting this going.

jerryneedell, 2018-03-23:
Just a quick update - Making progress on this. I have the basics working for SPI and I2C.  Patterned after FocalTouch as suggested and I have the basic "paint" demo working with a 2.4inch TFT. Lots of clean-up/tweaking to do.
Still have to deal with the occasional SPI "mode" issue. Seem like just retrying usually gets it "right".

ladyada, 2018-03-23:
sounds great!

jerryneedell, 2018-07-20:
done

closed 2018-07-20 by jerryneedell