adafruit/circuitpython #584: Validate high priority drivers for 3.0 release

tdicola, 2018-02-05:
This is an issue to track what drivers have been validated for the 3.0 release.  In discussion on 1/5/18 we decided this is an important release criteria to gate a 3.0 release--validating popular drivers that exercise many code paths.  In particular with 3.0 we have some big changes to memory management and numeric representation/range so validation of drivers that deal with those issues is a priority.

A few to consider:
- Basic I2C usage: Any simple BME/BMP pressure sensor.
- Basic SPI usage: MAX31865 thermocouple
- Basic bitbang / pulse IO: DHT sensors
- Popular: Featherwings.. ideally all of them, but OLED, CharliePlex are pretty popular and well used.
- Memory usage: Thermal printer
- Large numbers: SI5351 clock generator
- Other interesting ones: RFM69 radio

ladyada, 2018-02-05:
- Multiple SPI devices: TFT FeatherWing (my go-to demo is drawing a bitmap from disk to TFT but a 'touch paint' demo does STMPE + TFT)
- Clock-stretch I2C: BNO055
- framebuf: OLED
- Bi-directional UART: GPS with changing settings via NMEA sentences

dhalbert, 2018-06-17:
Did a bunch of testing (see checkboxes). Edited some of the tests above (no, I did not test all the Featherwings, just some representative ones0. @kattni: I think the CPX testing is basically done, right? Maybe we can close this.

tannewt, 2018-06-18:
Sounds like we can close this. @dhalbert want to do the honors?

kattni, 2018-06-18:
Was `pulsein` and `pulseout` tested on CPX?

dhalbert, 2018-06-18:
I'll test CPX IR with some existing Learn Guide code.

dhalbert, 2018-06-18:
All tested!

closed 2018-06-18 by dhalbert