adafruit/circuitpython #587: nRF52840: USB Support

microbuilder, 2018-02-06:
USB support should be added for the following classes as a minimum:

- [ ] USB CDC
- [ ] USB Mass Storage
- [ ] USB HID

A review is currently pending on the best solution for the USB stack, whether to use the Nordic provided stack or another open source alternative that can be used across platforms.

tannewt, 2018-02-06:
I'd love to be involved in the USB stack discussion. We're not very wedded to ASF4 at all for SAMD21 and SAMD51.

microbuilder, 2018-02-06:
I'd love to be able to get @hathach to spend a bit of time updating https://github.com/hathach/tinyusb since we'd have full control over the design decisions there, and it was designed from the start to be vendor neutral (which clearly isn't the case with most other stacks out there). Other options may make more sense after some digging, of course, but I think tinyusb is a strong option with a permissive license.

tannewt, 2018-02-06:
@microbuilder I'm on board with that. Do we expect much difference in timeline to get USB on nrf52840 going between the two options?

microbuilder, 2018-02-06:
I'll have to ping Thach for an estimate here since he's the one who will be doing the brunt of the USB work, since he knows it best.

microbuilder, 2018-02-22:
Now that @hathach is back from holidays (new year), I'm going to do some project planning to figure out the minimum requirements for a viable USB solution that covers both SAMDxx and nRF52840 and should be maintainable across future platforms. It will be a separate repo and project so that I can do the project management on Github and track things there, but I'll post a link here once a first pass at requirements and deliverables is done (on the USB side, not CP-specific at this point since it will also need to play well with the Arduino BSP), and we can extend the feedback loop from there.

tannewt, 2018-02-22:
@microbuilder sounds excellent. @dpgeorge may want to use it in MicroPython for nRF52 USB as well.

tannewt, 2018-07-10:
Split into #1007 #1008 and #1009

closed 2018-07-10 by tannewt