adafruit/circuitpython #590: nRF52: Clean up examples

arturo182, 2018-02-06:
It seems some of the examples in the nrf port are not valid anymore, for example the seeed tft one uses framebuf and machine, neither of those modules is available in the port.

Shouldn't the examples be better structured, the ones in the root directory should work on "all" supported ports, the ones inside the port directory should work on all boards that port supports and examples inside a board directory should be board specific and only work on that one (for example if some boards have buttons and others don't).

tannewt, 2018-02-06:
Yes! Examples in the repo aren't critical because we have detailed tutorials and how-tos on https://learn.adafruit.com.

tannewt, 2018-07-10:
This done?

arturo182, 2018-07-10:
No yet, waiting for the cleanup to be done so I can write proper examples and test them with the new code :)

tannewt, 2019-02-05:
@dhalbert has done this:
* https://github.com/adafruit/Adafruit_CircuitPython_BLE/
* https://github.com/adafruit/Adafruit_CircuitPython_BluefruitConnect
* https://learn.adafruit.com/circuspython-jump-through-hoops-with-bluetooth-le

closed 2019-02-05 by tannewt