adafruit/circuitpython #593: Allow ifdefing out classes in the analogio module

arturo182, 2018-02-06:
Some MCUs, like the nRF do not support analog output so it would be best to not have the AnalogOut class available on that port, right now it is no possible and the constructor just throws an exception, which will at least inform the user that they are trying to perform a unsupported operation.

tannewt, 2018-02-06:
I prefer throwing an exception instead of removing it. I only want to remove things at the module level, not the class level. We have a chart for module support here: https://circuitpython.readthedocs.io/en/latest/shared-bindings/index.html#support-matrix

I understand this costs a bit of code size but would prefer a specific error message over a name not found exception. The ESP port raises an OSError: https://github.com/adafruit/circuitpython/blob/master/ports/esp8266/common-hal/analogio/AnalogOut.c#L37

arturo182, 2018-02-06:
Ok, I see, I will update my PR to make it consistent with the ESP port.

closed 2018-02-07 by arturo182