adafruit/circuitpython #597: Support for 48-pin ATSAMD51G devices

siddacious, 2018-02-07:
There are a very small handful of places that assume that if you're using a SAMD51 that you're using a SAMD51J. We should support the SAMD51G to allow us to pack the SAMD51/CP wombo-combo into an even smaller package.

From looking at the datasheet, the differences between the 51J and 51G are
- Number of TCs
- Number of 16bit TCCs
- I2S support
- Number of GPIO
- Number of ADC channels on ADC1
- Number of PTC channels
- Number of tamper pins

The only problems I've found so far are

- samd51_pins.c reflects a confusion in the datasheet about TC4 existing on the SAMD51G chips. We should add a check for it. There isn't any confusion in the datasheet about the TCCs or ADCs so those are fine here
 
- ports/atmel-samd/supervisor/port.c will output GCLK0 and GLK1 on some pins if you're on a SAMD51, however the pins chosen don't exist on the SAMD51G. On my branch I have the lines in question changed to new pins that are common between the two, but we can either do that or have some logic looking at MICROPY_HW_MCU_NAME perhaps to choose pins depending on the chip.

I have the fixes on a branch and can submit a PR once we decide what to do about the clock debug pins. If I find any other issues I can open a new issue.

tannewt, 2018-02-07:
I'm on board with supporting the samd51G.

For the clock debug pins we can either change them or simply remove them. Its there for debugging only and can be added back as needed.

PRs welcome! Thank you!

ladyada, 2018-02-08:
fyi feather will be 51g 😀

siddacious, 2018-02-08:
@ladyada 👍 
Let me know if you need alpha/beta testers!

tannewt, 2018-02-13:
This done @siddacious ?

siddacious, 2018-02-13:
It's compiling, loading and giving me a REPL and a working status neopixel however in testing I ran into some SPI issues but I don't know how to differentiate problems with my board and board config from CP issues.

 I think it's good enough for now to allow others to build for a 51G and get others using it. I'll open another issue if I find something that needs to be fixed.

closed 2018-02-13 by siddacious