adafruit/circuitpython #605: WebUSB REPL

ladyada, 2018-02-08:
https://developers.google.com/web/updates/2016/03/access-usb-devices-on-the-web
https://github.com/webusb/arduino
Expose REPL w/WebUSB HID -> no native serial monitor required whatsoever
**But** is Chrome only right now. Still, Chrome is a popular browser.
Lets ponder for 4.0 or beyond!

tannewt, 2018-02-08:
I think this is a good idea too! We could share a lot of development work between a WebUSB webpage and a standalone, cross-platform Electron app (which is chromium under the hood).

theacodes, 2019-11-11:
I would love to look into this. I'm adding it to my list of things to look into while I have free time over the holidays, but it should absolutely not discourage anyone else from taking this on - just please reach out if you want to collaborate. :)

ladyada, 2019-11-11:
i thiiiiiiiiiiiink we needed more USB endpoints? or were waiting till we could config the USB endp on/off states - e.g. disable MSD on demand

ladyada, 2019-11-12:
note there's webserial nowadays. could be interesting to try that - it may 'just work' (?)

theacodes, 2019-11-12:
That's in interesting option as well.

ladyada, 2020-02-03:
following up, with WebSerial now included, this is not important - we can use the every day repl in a browser! it would be nice to have webusb eventually but i don care too much anymore :)

theacodes, 2020-02-03:
Huzzah!

carlosperate, 2020-02-27:
>I think this is a good idea too! We could share a lot of development work between a WebUSB webpage and a standalone, cross-platform Electron app (which is chromium under the hood).

I know this is a very old comment, but just in case this information is useful, the device chooser for electron has not been implemented yet, and I assume this would also apply for WebSerial: https://github.com/electron/electron/issues/14545
But, I think this node implementation could be used as a WebUSB polyfill: https://github.com/thegecko/webusb

ladyada, 2020-11-19:
closing this cause we can just use webserial instead of a webusb

closed 2020-11-19 by ladyada