adafruit/circuitpython #613: Add/fix ReadTheDocs badge

tannewt, 2018-02-12:
We had a bad cookiecutter for a while that added extra lines into the `README.rst` in the ReadTheDocs badge like this:

```rst
.. image:: https://readthedocs.org/projects/adafruit-circuitpython-vcnl4010/badge/?version=latest

    :target: https://circuitpython.readthedocs.io/projects/vcnl4010/en/latest/

    :alt: Documentation Status
```

It should be:

```rst
.. image:: https://readthedocs.org/projects/adafruit-circuitpython-vcnl4010/badge/?version=latest
    :target: https://circuitpython.readthedocs.io/projects/vcnl4010/en/latest/
    :alt: Documentation Status
```

These are super simple changes that are great practice for using GitHub. Feel free to ask questions here as needed.

These repos need a look:

- [ ] https://github.com/adafruit/Adafruit_CircuitPython_VCNL4010
- [ ] https://github.com/adafruit/Adafruit_CircuitPython_Fingerprint
- [ ] https://github.com/adafruit/Adafruit_CircuitPython_APDS9960
- [ ] https://github.com/adafruit/Adafruit_CircuitPython_DRV2605
- [ ] https://github.com/adafruit/Adafruit_CircuitPython_SGP30
- [ ] https://github.com/adafruit/Adafruit_CircuitPython_DS18X20
- [ ] https://github.com/adafruit/Adafruit_CircuitPython_AVRprog
- [ ] https://github.com/adafruit/Adafruit_CircuitPython_FocalTouch
- [ ] https://github.com/adafruit/Adafruit_CircuitPython_MCP9808
- [ ] https://github.com/adafruit/Adafruit_CircuitPython_VL6180X
- [ ] https://github.com/adafruit/Adafruit_CircuitPython_OneWire
- [ ] https://github.com/adafruit/Adafruit_CircuitPython_Thermal_Printer
- [ ] https://github.com/adafruit/Adafruit_CircuitPython_DS2413
- [ ] https://github.com/adafruit/Adafruit_CircuitPython_VEML6070
- [ ] https://github.com/adafruit/Adafruit_CircuitPython_MAX9744
- [ ] https://github.com/adafruit/Adafruit_CircuitPython_MCP4725
- [ ] https://github.com/adafruit/Adafruit_CircuitPython_SSD1306
- [ ] https://github.com/adafruit/Adafruit_CircuitPython_SHT31D
- [ ] https://github.com/adafruit/Adafruit_CircuitPython_TLC5947
- [ ] https://github.com/adafruit/Adafruit_CircuitPython_Register

Part of #566

brentru, 2018-02-14:
@tannewt Note - the [SSD1306 repo](https://github.com/adafruit/Adafruit_CircuitPython_SSD1306) 404's [on its ReadTheDocs page](https://circuitpython.readthedocs.io/projects/ssd1306/en/latest/)

tannewt, 2018-02-14:
Thats the right link for ssd1306 but the build itself isn't set up right. The build output is here: https://readthedocs.org/projects/adafruit-circuitpython-ssd1306/builds/6661205/

I'd expect register to be at https://circuitpython.readthedocs.io/projects/register/en/latest/

mrmcwethy, 2018-02-20:
@ tannewt @brentru I believe this issue can now be closed.

tannewt, 2018-02-20:
Great job @mrmcwethy and @brentru !

closed 2018-02-20 by tannewt