adafruit/circuitpython #616: I2C issue on NRF52 (PCA10056 board)

jerryneedell, 2018-02-12:
I have been seeing an issue with I2C on the PCA10056 board with the CircuitPython master as of this morning: 
```
Adafruit CircuitPython 3.0.0-alpha.1-140-g0388a57 on 2018-02-12; PCA10056 with NRF52840
```

I am opening this as a "heads up" and I'll continue to experiment with it. 
The error reported below is using the vl53l0x driver from the current "bundle" and the test code from the examples subdirectory of the repo
https://github.com/adafruit/Adafruit_CircuitPython_VL53L0X

for this case I have SCL on PA27 and SDA on PA26

```
# Simple demo of the VL53L0X distance sensor.
# Will print the sensed range/distance every second.
import time

import board
import busio

import adafruit_vl53l0x

# Initialize I2C bus and sensor.
i2c = busio.I2C(board.PA27, board.PA26)
vl53 = adafruit_vl53l0x.VL53L0X(i2c)

# Optionally adjust the measurement timing budget to change speed and accuracy.
# See the example here for more details:
#   https://github.com/pololu/vl53l0x-arduino/blob/master/examples/Single/Single.ino
# For example a higher speed but less accurate timing budget of 20ms:
#vl53.measurement_timing_budget = 20000
# Or a slower but more accurate timing budget of 200ms:
#vl53.measurement_timing_budget = 200000
# The default timing budget is 33ms, a good compromise of speed and accuracy.

# Main loop will read the range and print it every second.
while True:
    print('Range: {0}mm'.format(vl53.range))
    time.sleep(1.0)

```

```
>>> import vl53l0x_example
Range: 8190mm
Range: 8190mm
Range: 141mm
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "vl53l0x_example.py", line 25, in <module>
  File "adafruit_vl53l0x.py", line 498, in range
  File "adafruit_vl53l0x.py", line 291, in _read_u8
  File "adafruit_vl53l0x.py", line 290, in _read_u8
  File "adafruit_bus_device/i2c_device.py", line 102, in write
TypeError: extra keyword arguments given
>>> 
```

I can't make a lot of sense out of the traceback or the error being thrown.

Note - This same code (with the default SCL,SDA)runs fine on my feather52 board.

arturo182, 2018-02-12:
Thanks for the heads up, I have not tested the DKs with I2C at all so it's quite possible it's broken there. Tomorrow I'll grab some I2C boards from my hackerspace to test, if I can reproduce and get a backtrace in C then it will be easier to fix.

jerryneedell, 2018-02-12:
@arturo182 FYI - I have not seen any I2C errors with an mcp9808 temperature humidity sensor. I'll try a few more sensors and report.

arturo182, 2018-07-17:
@jerryneedell could you retest this now that we use nRFx for I2C?

jerryneedell, 2018-07-18:
sure, will do it as soon as I can.

jerryneedell, 2018-07-19:
Rested and the vl53l0x_example runs as expected -- closing this issue!

closed 2018-07-19 by jerryneedell