adafruit/circuitpython #619: asyncio, async, and await would be super helpful

wolf, 2018-02-12:
`asyncio` provides event loop code and asynchronous coroutines that fit perfectly with the I/O needs of these all-I/O boards.

Strongly related to <https://github.com/adafruit/circuitpython/issues/197>, though things have changed since then.  I'd love to see `async` and `await` available and used with `asyncio`.  I have time to work on this but would need guidance on how to do the right thing.

tannewt, 2018-02-13:
@wolf adding this would be awesome! I don't know much about asyncio myself so it'd be great to chat about what is needed for it. I know micropython has uasyncio but haven't looked at it myself. I'd want us to match the way CPython works even if MicroPython doesn't.

When are you available to chat or what guidance can I help with? I'm on Discord during the days Pacific time and can hop on other times if I know ahead of time. Thanks!

kattni, 2018-02-13:
It'd be great if we could coordinate with @dhalbert as well.

wolf, 2018-02-13:
Telling me that we want to match CPython over MicroPython is exactly the kind of guidance I'm looking for.  I'll have to figure out the details; I'm starting from scratch here.  Anything you can tell me about goals, constraints and dreams will be a help.  I am Eastern time, so my 4pm (about when I arrive home from work everyday) might be convenient for both of us.  Once we pick a time, just about any day works for me.  Work for you? (and @dhalbert and @kattni)

tannewt, 2018-02-13:
4pm eastern works great for me! Wednesday - Friday are best for me.

There is more design guidance here: https://circuitpython.readthedocs.io/en/latest/docs/design_guide.html Its a random assortment of stuff.

dhalbert, 2018-02-13:
Yes, most days are fine for me. Just Monday afternoons are taken for all three of us. You saw that there is a `MICROPY_PY_ASYNC_AWAIT` compile option? We just don't have it turned on.

wolf, 2018-02-13:
Then I propose this coming Thursday, 15 February at 4pm Eastern on Discord.  That will give me a little time to do research first so I have the right grounding to absorb your advice.

kattni, 2018-02-13:
Thursday 4pm works for me.

tannewt, 2018-02-13:
Works for me!

dhalbert, 2018-02-13:
OK by me too.

sander76, 2018-04-04:
Any news on this ? Starting to jump in circuitpython and I can see the benefits here. (Having experience with asyncio on Cpython)

wolf, 2018-04-04:
My goal is to have this working for PyCon and I’m still on track for that.  As soon as I get something that doesn’t break the world I’ll push a branch for people to play with.

notro, 2019-01-10:
See #1415 for some work on this.

dhalbert, 2020-11-01:
Fixed by #3540.

closed 2020-11-01 by dhalbert