adafruit/circuitpython #620: Absolute mouse (touchscreen) HID support

ladyada, 2018-02-13:
Good one to do when we do multimedia key support (i believe we just need to craft the report API)

dhalbert, 2018-04-09:
I did a little research on this. It may be an issue finding report descriptors that will work on all the major platforms. Paul Stoffregen has implemented touchscreen HID support for Teensy. The report descriptor he developed I think appears to work on Windows and Mac without external drivers; it may not work on Linux (or at least Raspbian). Some background here and in other forum posts: https://forum.pjrc.com/threads/32331-USB-HID-Touchscreen-support-needed

dhalbert, 2018-04-24:
Latent support in #776, but not turned on. Needs more testing: it appears touchscreen and gamepad on the same HID endpoints might conflict on at least some platforms (Linux).  Might need to wait for dynamic HID descriptors. Also current report descriptor (emulating a digitizer with stylus), does not necessarily seem to do the right thing with existing built-in drivers on all platforms.

dhalbert, 2021-06-15:
Now available via user-defined HID descriptors.

closed 2021-06-15 by dhalbert

ladyada, 2021-06-16:
woohoo 3 year old issue :D

ATMakersBill, 2021-06-16:
While I understand how this solves the problem in theory, is there an example or doc that has the descriptor and hid code to process it?

dhalbert, 2021-06-16:
Digitizer descriptor that worked on at least some systems:

https://github.com/adafruit/circuitpython/blob/6.3.x/tools/hid_report_descriptors.py#L341-L406

I don't know what Python library I was using with this, if any. I may have just seen whether the OS recognized it, as opposed to trying to send a report.

Red-M, 2021-06-16:
For extra compatibility, these are the USB descriptors that QEMU/KVM use, https://github.com/qemu/qemu/blob/master/hw/usb/dev-hid.c#L398-L410

bitboy85, 2021-07-31:
If anyone ends up here by google: I've created an example for absolute mouse here: https://gist.github.com/bitboy85/cdcd0e7e04082db414b5f1d23ab09005

hdo, 2021-11-26:
Anyone have a working example with CircuitPython 7.0? 
Unfortunately i can't get it working with CircuitPython 6.3 on PICO :-(

ThomasAtBBTF, 2021-11-26:
I have one and will upload it here over the weekend.
BUT: It works on MacOS but not on IOS !
Did not check Windows and Linux.
I opened a feature request to Apple for making it work also under IOS.

hdo, 2021-11-26:
That would be very helpful. Thanks.

> I have one and will upload it here over the weekend. BUT: It works on MacOS but not on IOS ! Did not check Windows and Linux. I opened a feature request to Apple for making it work also under IOS.