adafruit/circuitpython #622: Consistency in unified sensor driver property naming

tuupola, 2018-02-13:
To avoid hijacking an [unrelated thread](https://github.com/adafruit/circuitpython/pull/600) I move this here. Currently there is [mixed capitalization](https://github.com/tannewt/circuitpython/blob/09bf06bc6bf4eba0ad03916425ad45ebab9f0025/docs/design_guide.rst#L470-L472) in the design guide concerning unified sensor driver property naming.

Consistency is usually good thing. If everything is lowercased there is no need to remember whether one should use capital letters or not. For example lux might be confusing. Is it acronym or word? Should it be `sensor.lux` or `sensor.LUX`? Also when seeing `sensor.TVOC` one is not really sure if it is a constant or property.

I do not have strong feelings about this, but my choice would be lowercased acronyms `sensor.tvoc` and `sensor.eco2`.

tannewt, 2018-02-15:
I've been silent on this because I haven't felt super strong about it. I thought about expanding them out (`total_volatile_organic_compound` for example) but then they get really long. Ultimately, I agree with you @tuupola. Switching to lowercase makes sense to me.

tannewt, 2018-02-28:
@tuupola Would you like to switch them to the lower case version?

brentru, 2018-03-05:
These two libs to be changed to reflect the move to lowercase (am I forgetting any others which use the TVOC/eCO2 property naming?):
https://github.com/adafruit/Adafruit_CircuitPython_SGP30
https://github.com/adafruit/Adafruit_CircuitPython_CCS811

tuupola, 2018-03-05:
Had missed this. @tannewt sure, I will check them out.

tannewt, 2018-03-05:
Thanks @tuupola! Let me know if you need any guidance on how to get it done.

dherrada, 2018-05-14:
@tuupola I put in a pull request to the [CCS811](https://github.com/adafruit/Adafruit_CircuitPython_CCS811) library and it was accepted. I didn't see any problems with the SGP30 library.

tannewt, 2018-05-16:
SGP30 is missing properties all together. They should be added.

tannewt, 2018-05-25:
@mrmcwethy would you be interested in adding eco2 and tvoc properties to the SGP30 driver?

mrmcwethy, 2018-05-29:
@tannewt I would be happy to add these properies.  I will need to order a sgp30 board.

tannewt, 2018-05-29:
Go ahead and order one (and other other sensors of interest). I think @brentru may have started the sgp30 work though.

tannewt, 2018-05-29:
Yes please. I'd name them `baseline_tvoc` and `baseline_co2eq` though. Thanks!

tannewt, 2018-06-01:
The CCS guide needs to be updated: https://learn.adafruit.com/adafruit-ccs811-air-quality-sensor?view=all#circuitpython-wiring-test

brentru, 2018-06-03:
@tannewt  will do.

brentru, 2018-06-04:
Modified the CCS and SGP30 guides to match the naming consistencies.

tannewt, 2018-06-04:
Awesome! I think this is done then. Please reopen if not.

closed 2018-06-04 by tannewt