adafruit/circuitpython #624: Fix autodoc in library doc builds

tannewt, 2018-02-15:
These repos have docs that may technically pass but the autodoc portion fails. This is usually due to a library not being mocked out. To fix it, a line must be added to conf.py to tell sphinx to mock out specific modules. Here is the line from the latest cookicutter's conf.py:

```py
# Uncomment the below if you use native CircuitPython modules such as
# digitalio, micropython and busio. List the modules you use. Without it, the
# autodoc module docs will fail to generate with a warning.
# autodoc_mock_imports = ["adafruit_bus_device", "micropython"]
```

Copy these four lines, uncomment the fourth and adjust the list until the sphinx build includes the autodocs.

- [ ] https://github.com/adafruit/Adafruit_CircuitPython_INA219
- [ ] https://github.com/adafruit/Adafruit_CircuitPython_DS3231
- [ ] https://github.com/adafruit/Adafruit_CircuitPython_CharLCD
- [ ] https://github.com/adafruit/Adafruit_CircuitPython_MAX31865
- [ ] https://github.com/adafruit/Adafruit_CircuitPython_AMG88xx
- [ ] https://github.com/adafruit/Adafruit_CircuitPython_VL53L0X
- [ ] https://github.com/adafruit/Adafruit_CircuitPython_TCS34725
- [ ] https://github.com/adafruit/Adafruit_CircuitPython_seesaw
- [ ] https://github.com/adafruit/Adafruit_CircuitPython_LSM9DS1
- [ ] https://github.com/adafruit/Adafruit_CircuitPython_LSM9DS0
- [ ] https://github.com/adafruit/Adafruit_CircuitPython_Thermistor
- [ ] https://github.com/adafruit/Adafruit_CircuitPython_VC0706
- [ ] https://github.com/adafruit/Adafruit_CircuitPython_HT16K33
- [ ] https://github.com/adafruit/Adafruit_CircuitPython_FXOS8700
- [ ] https://github.com/adafruit/Adafruit_CircuitPython_FXAS21002C
- [ ] https://github.com/adafruit/Adafruit_CircuitPython_LIS3DH
- [ ] https://github.com/adafruit/Adafruit_CircuitPython_RTTTL
- [ ] https://github.com/adafruit/Adafruit_CircuitPython_IS31FL3731
- [ ] https://github.com/adafruit/Adafruit_CircuitPython_ADS1x15
- [ ] https://github.com/adafruit/Adafruit_CircuitPython_LSM303
- [ ] https://github.com/adafruit/Adafruit_CircuitPython_SimpleIO
- [ ] https://github.com/adafruit/Adafruit_CircuitPython_BNO055
- [ ] https://github.com/adafruit/Adafruit_CircuitPython_VCNL4010
- [ ] https://github.com/adafruit/Adafruit_CircuitPython_Fingerprint
- [ ] https://github.com/adafruit/Adafruit_CircuitPython_SGP30
- [ ] https://github.com/adafruit/Adafruit_CircuitPython_SHT31D
- [ ] https://github.com/adafruit/Adafruit_CircuitPython_APDS9960
- [ ] https://github.com/adafruit/Adafruit_CircuitPython_DRV2605
- [ ] https://github.com/adafruit/Adafruit_CircuitPython_DS2413
- [ ] https://github.com/adafruit/Adafruit_CircuitPython_OneWire
- [ ] https://github.com/adafruit/Adafruit_CircuitPython_VEML6070
- [ ] https://github.com/adafruit/Adafruit_CircuitPython_VL6180X
- [ ] https://github.com/adafruit/Adafruit_CircuitPython_Thermal_Printer
- [ ] https://github.com/adafruit/Adafruit_CircuitPython_FocalTouch
- [ ] https://github.com/adafruit/Adafruit_CircuitPython_MAX9744
- [ ] https://github.com/adafruit/Adafruit_CircuitPython_MCP4725
- [ ] https://github.com/adafruit/Adafruit_CircuitPython_AVRprog
- [ ] https://github.com/adafruit/Adafruit_CircuitPython_DS18X20

sommersoft, 2018-02-16:
Just for clarification/get-out-in-front of questions: will these need PRs to the bundle repo, or will adabot pull them automagically?

tannewt, 2018-02-16:
Just PRs to individual repos. ReadTheDocs is based on each repo so it'll pick them up once merged. They don't even need a release.

tannewt, 2018-03-19:
https://github.com/adafruit/Adafruit_CircuitPython_APDS9960/pull/8 is the last fix for this I believe.

sommersoft, 2018-04-30:
Belated close-n-thanks, @brentru!!!

closed 2018-04-30 by sommersoft