adafruit/circuitpython #637: Error building 2.2.3 for HUZZAH

deshipu, 2018-02-23:
I'm trying to build my own version on the 2.2.3 CircuitPython for HUZZAH, but I'm getting the following error:
```
In file included from ./esp_mphal.h:73:0,
                 from ../py/mphal.h:32,
                 from ../py/mpprint.c:33:
./etshal.h:9:6: error: conflicting types for 'ets_delay_us'
 void ets_delay_us();
      ^
./etshal.h:9:1: note: an argument type that has a default promotion can't match an empty parameter name list declaration
 void ets_delay_us();
 ^
In file included from ./esp_mphal.h:66:0,
                 from ../py/mphal.h:32,
                 from ../py/mpprint.c:33:
/home/sheep/dev/micropython/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/include/osapi.h:33:6: note: previous declaration of 'ets_delay_us' was here
 void ets_delay_us(uint16_t us);
      ^
../py/mkrules.mk:53: recipe for target 'build/py/mpprint.o' failed
make: *** [build/py/mpprint.o] Error 1
```
It seems like the esp-open-sdk that is mentioned in the README is no longer compatible? What version of esp-open-sdk should I be using?

deshipu, 2018-02-23:
The following changes made it compile:

```diff
diff --git a/esp8266/etshal.h b/esp8266/etshal.h
index 34787779f..32942bcb2 100644
--- a/esp8266/etshal.h
+++ b/esp8266/etshal.h
@@ -6,14 +6,14 @@
 // see http://esp8266-re.foogod.com/wiki/Random_Number_Generator
 #define WDEV_HWRNG ((volatile uint32_t*)0x3ff20e44)
 
-void ets_delay_us();
+void ets_delay_us(uint16_t us);
 void ets_intr_lock(void);
 void ets_intr_unlock(void);
 void ets_isr_mask(uint32_t mask);
 void ets_isr_unmask(uint32_t mask);
 void ets_isr_attach(int irq_no, void (*handler)(void *), void *arg);
 void ets_install_putc1();
-void uart_div_modify();
+void uart_div_modify(uint8 uart_no, uint32 DivLatchValue);
 void ets_set_idle_cb(void (*handler)(void *), void *arg);
 
 void ets_timer_arm_new(os_timer_t *tim, uint32_t millis, bool repeat, bool is_milli_timer);
@@ -33,10 +33,10 @@ void MD5Update(MD5_CTX *context, const void *data, unsigned int len);
 void MD5Final(unsigned char digest[16], MD5_CTX *context);
 
 // These prototypes are for recent SDKs with "malloc tracking"
-void *pvPortMalloc(unsigned sz, const char *fname, int line);
-void *pvPortZalloc(unsigned sz, const char *fname, int line);
-void *pvPortRealloc(void *p, unsigned sz, const char *fname, int line);
-void vPortFree(void *p, const char *fname, int line);
+void *pvPortMalloc (size_t sz, const char *, unsigned);
+void *pvPortZalloc (size_t sz, const char *, unsigned);
+void *pvPortRealloc (void *p, size_t n, const char *, unsigned);
+void vPortFree (void *p, const char *, unsigned);
 
 uint32_t SPIRead(uint32_t offset, void *buf, uint32_t len);
 uint32_t SPIWrite(uint32_t offset, const void *buf, uint32_t len);
```

deshipu, 2018-02-23:
Should I make a pull request?

deshipu, 2018-02-23:
Even after that, I still get:
```
xtensa-lx106-elf-ld: build/firmware.elf section `.text' will not fit in region `iram1_0_seg'
xtensa-lx106-elf-ld: region `iram1_0_seg' overflowed by 61096 bytes
```
I guess the firmware grew too much?

jerryneedell, 2018-02-23:
hmmm  - I just did a fresh pull and checkout of 2.x and I get the following  error:
```
make[2]: Leaving directory '/home/jerryneedell/projects/adafruit_github/circuitpython_2x/lib/axtls/ssl'
make[1]: Leaving directory '/home/jerryneedell/projects/adafruit_github/circuitpython_2x/lib/axtls'
cp ../lib/axtls/_stage/libaxtls.a build/libaxtls.a
make: *** No rule to make target 'drivers/dht/dht.c', needed by 'build/genhdr/qstr.i.last'.  Stop.
```

jerryneedell, 2018-02-23:
It looks like the drivers/dht folder was removed  with #632 - this breaks esp8266 build for 2.x as noted above.

jerryneedell, 2018-02-23:
@deshipu my version of the SDK is pretty old - I have not updated it recent;y - perhaps I'll hold off...

deshipu, 2018-02-23:
@jerryneedell you can always checkout an earlier version, the thing is, we need to know which one...

jerryneedell, 2018-02-23:
@deshipu how can I verify the version? I also see 
ESP8266_NONOS_SDK_V2.0.0_16_08_10
in my build folder.

jerryneedell, 2018-02-23:
looks like it is:
sdk -> ESP8266_NONOS_SDK_V2.0.0_16_08_10

deshipu, 2018-02-23:
Oh, os it's the pre-krak one.

jerryneedell, 2018-02-23:
It is the same version on my linux box (May 2017) and on a "vagrant" install I did on my Mac  in January. I have not updated it since then. It looks like the vagrant install used the latest available at the time.

jerryneedell, 2018-02-23:
does your 2.x clone have the "drivers/dht "folder ?

deshipu, 2018-02-23:
No `drivers` in 2.x

jerryneedell, 2018-02-23:
OK so do you also get the same error I am getting from the Makefile? Or don't you get that far.

jerryneedell, 2018-02-23:
[https://github.com/adafruit/circuitpython/blob/2.x/esp8266/Makefile#L193](url)

deshipu, 2018-02-23:
For now I'm trying to build 2.2.3, I didn't try with 2.x, since I have some modifications I wanted to test that I don't want to commit.

deshipu, 2018-02-23:
I think that your issue is separate from what I'm getting.

jerryneedell, 2018-02-23:
@deshipu yes - sorry if I hijacked your issue - should I create a new one?

deshipu, 2018-02-23:
I think it would be best for keeping track of both of them in a non-confusing way :)

dhalbert, 2018-02-23:
@deshipu that is a big increase in size (60k) - we didn't add _that_ much. PR's to fix would be great.
@jerryneedell I see the dht issue. We should remove that driver completely, since we have a CircuitPython substitute. You can open a quick issue to remind us.

deshipu, 2018-02-23:
It seems like the increase in size is also due to the new SDK. Checking out commit 4ea62358c98799c1d4d1e711e5cff08045fc4dec and building with that makes it work.

deshipu, 2018-02-23:
The change looks innocent... https://github.com/pfalcon/esp-open-sdk/commit/5518fb6116c35a02ccb9a87260bb194a57cb429e

deshipu, 2018-02-23:
The change for esthal.h is here: https://github.com/pfalcon/micropython/commit/e766a4af4a336816d42d65c01ee6c3e4cc207b7e

deshipu, 2018-02-23:
I think we should just come up with some way of making sure that a given version of CircuitPython gets built with a matching version of the esp-open-sdk. There are going to be more breaking changes like this one, and it's not feasible for us to keep updating all the branches to fix them.

tannewt, 2018-07-03:
@jepler Did you do this?

jepler, 2018-07-04:
@tannewt at 52c0fc421b241ae581b9c69f45cfd5bd50289ed6 we switched to building (3.0 and newer) on travis with a specific sdk build.  However, I didn't update documentation to mention anything about this.  I'm not sure what to write that will best help user-developers.  Hopefully, CP doesn't fall that far behind the tip of esp-open-sdk (there have been no commits there since the release I made on 2016-06-10), in which case the existing documentation is perfectly adequate.  I'm fine if you want to close this up.

tannewt, 2018-07-05:
@jepler Mind updating this? https://github.com/adafruit/circuitpython/blob/master/ports/esp8266/README.md#build-instructions

tannewt, 2018-10-09:
This is done. I won't block this on the documentation.

closed 2018-10-09 by tannewt