adafruit/circuitpython #642: Can't use GPIO12 as DigitalInOut on ESP8266

deshipu, 2018-02-24:
Trying to initialize DigitalInOut on pin GPIO12 results in:

```
>>> dc = digitalio.DigitalInOut(board.GPIO12)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: Pin MTDI in use
```

deshipu, 2018-02-24:
Seems like the pins are not reset on reset.

deshipu, 2018-02-24:
I did some more digging, and the issue is actually more complex. It looks like you can't use GPIO12 whenever the SPI perihpheral is created, even if you only defined MOSI and no MISO for it.

tannewt, 2018-02-27:
@deshipu Could you provide the full test code for the after-SPI init problem you are seeing? Thanks!

tannewt, 2018-05-23:
@sommersoft Does this sound familiar to you?

sommersoft, 2018-05-23:
No, it doesn't. I know there are about 6 special pins, but 12 doesn't ring a bell as being one. 0,2,5,6,15, and 16 are the ones I remember off the top of my head. But, I can peek at it a little later.

Initial thought is that there are default settings that could be tweaked in SPI...

sommersoft, 2018-05-23:
Pretty confident I found it. 

[`common_hal_busio_spi_configure`](https://github.com/adafruit/circuitpython/blob/master/ports/esp8266/common-hal/busio/SPI.c#L109) Lines 109 and 121 both call [`hspi.c/spi_init_gpio()`](https://github.com/adafruit/circuitpython/blob/master/ports/esp8266/hspi.c#L92).

 `spi_init_gpio()` uses all 4 SPI pins (12-15) as a default. There is no conditional check...

So, we can either update `hspi.c`, or re-create the config steps in `common-hal` which is what the `construct` has done. I assume `common-hal` option is preferred, to maintain upstream compatible.

sommersoft, 2018-05-23:
Good catch. I think it's worth it to explore handling the possibilities... ?

commit 23b6d334 (2018-06-16): fixes #642; spi_init_gpio() is now bypassed for both construct and configure
closed 2018-06-16 by dhalbert