adafruit/circuitpython #644: Add module for non-blocking timer class

mikepschneider, 2018-02-25:
Hello,
I've created a simple non-blocking timer class for circuitpython using the time.monotonic() function.  It was suggested on discord that the class might be a useful addition to the circuitpython library for others to use.   If there's interest I'm happy to break out the class into a proper package.

Class is here:
https://github.com/mikepschneider/circuitpy_ms/blob/master/CIRCUITPY/nonblockingtimer.py

Example usages are here:
https://github.com/mikepschneider/circuitpy_ms/blob/master/CIRCUITPY/demos.py

mikepschneider, 2018-02-27:
OK I'm happy to help in any way but I'm new to this community so I don't know how things work.  If there's a next step I can/should do please let me know.

wolf, 2018-02-27:
I'm new to this community, too :-)  I don't know yet what the next step is for your code but it sounds like there may well be a place for it here.  I'll check it out.  Do you get on the Discord chat?  You and I should find a way to chat to figure out the next steps (probably with some direction from @tannewt and others).

mikepschneider, 2018-02-27:
Yes I'm on discord now and then.

tannewt, 2018-02-27:
Hi @mikepschneider Please see https://learn.adafruit.com/creating-and-sharing-a-circuitpython-library?view=all for instructions on structuring a CircuitPython library. Once you have it in a repo following that structure we can add it to the community bundle. Feel free to ping me on [Discord](https://adafru.it/discord) if you have questions.

mikepschneider, 2018-04-09:
I've started creating a repo here: https://github.com/mikepschneider/CircuitPython_nonblocking_timer.  Will update when there's something useful there to check out.

mikepschneider, 2018-05-21:
I've updated the nonblocking_timer code according to the CircuitPython library doc from above.  

Unit tests are running on travis-ci, but build is failing when building the circuitpython bundle.
See https://travis-ci.org/mikepschneider/CircuitPython_nonblocking_timer Could use some eyeballs on that @tannewt 

```$ circuitpython-build-bundles --filename_prefix circuitpython-nonblocking_timer --library_location .
/home/travis/build/mikepschneider/CircuitPython_nonblocking_timer
Multiple top level py files not allowed. Please put them in a package or combine them into a single file.
Generating VERSIONS
WARNING: some failures above
The command "circuitpython-build-bundles --filename_prefix circuitpython-nonblocking_timer --library_location ." exited with 2.
2.32s```

tannewt, 2018-05-22:
@mikepschneider I'd recommend moving the test file into a `test` directory. We may need to exempt a test directory from the build [here](https://github.com/adafruit/circuitpython-build-tools/blob/master/circuitpython_build_tools/build.py#L101) like we do for examples.

tannewt, 2018-07-03:
This is done: https://github.com/adafruit/CircuitPython_Community_Bundle/tree/master/libraries/helpers

closed 2018-07-03 by tannewt