adafruit/circuitpython #653: Cannot upload arduino sketch to meto_m4_rexpress_revb if CP3.0 executing

jerryneedell, 2018-03-01:
I am unable to upload an arduino sketch to my metro_m4_express_revb if I have previously loaded CP 3.0 to it and CP 3.0 is executing
I can upload successfully if I "double-tap" to put the M4 into the boot loader.
After loading an arduino sketch, I can load additional arduino sketches without first entering the boot loader.
I only have to manually enter the boot loader if CP 3.0 is loaded and executing.

Here is the entire build/load log for the blink.ino example  -  only the last bit is relevant, butI am including it all incase it is useful.
```
Arduino: 1.8.5 (Linux), Board: "Adafruit Metro M4 (SAMD51)"

/home/jerryneedell/arduino-1.8.5/arduino-builder -dump-prefs -logger=machine -hardware /home/jerryneedell/arduino-1.8.5/hardware -hardware /home/jerryneedell/.arduino15/packages -hardware /home/jerryneedell/Arduino/hardware -tools /home/jerryneedell/arduino-1.8.5/tools-builder -tools /home/jerryneedell/arduino-1.8.5/hardware/tools/avr -tools /home/jerryneedell/.arduino15/packages -built-in-libraries /home/jerryneedell/arduino-1.8.5/libraries -libraries /home/jerryneedell/Arduino/libraries -fqbn=Adafruit:samd:adafruit_metro_m4 -ide-version=10805 -build-path /tmp/arduino_build_551525 -warnings=none -build-cache /tmp/arduino_cache_975454 -prefs=build.warn_data_percentage=75 -verbose /home/jerryneedell/arduino-1.8.5/examples/01.Basics/Blink/Blink.ino
/home/jerryneedell/arduino-1.8.5/arduino-builder -compile -logger=machine -hardware /home/jerryneedell/arduino-1.8.5/hardware -hardware /home/jerryneedell/.arduino15/packages -hardware /home/jerryneedell/Arduino/hardware -tools /home/jerryneedell/arduino-1.8.5/tools-builder -tools /home/jerryneedell/arduino-1.8.5/hardware/tools/avr -tools /home/jerryneedell/.arduino15/packages -built-in-libraries /home/jerryneedell/arduino-1.8.5/libraries -libraries /home/jerryneedell/Arduino/libraries -fqbn=Adafruit:samd:adafruit_metro_m4 -ide-version=10805 -build-path /tmp/arduino_build_551525 -warnings=none -build-cache /tmp/arduino_cache_975454 -prefs=build.warn_data_percentage=75 -verbose /home/jerryneedell/arduino-1.8.5/examples/01.Basics/Blink/Blink.ino
Using board 'adafruit_metro_m4' from platform in folder: /home/jerryneedell/Arduino/hardware/Adafruit/samd
Using core 'arduino' from platform in folder: /home/jerryneedell/Arduino/hardware/Adafruit/samd
Detecting libraries used...
"/home/jerryneedell/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m4 -mthumb -c -g3 -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions  -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10805 -DARDUINO_METRO_M4 -DARDUINO_ARCH_SAMD  -D__SAMD51J19A__ -D__SAMD51__ -DUSB_VID=0x239A -DUSB_PID=0x8020 -DUSBCON '-DUSB_MANUFACTURER="Adafruit LLC"' '-DUSB_PRODUCT="Adafruit Metro M4"' -D__FPU_PRESENT -DARM_MATH_CM4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 "-I/home/jerryneedell/.arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/" "-I/home/jerryneedell/.arduino15/packages/arduino/tools/CMSIS-Atmel/1.1.0/CMSIS/Device/ATMEL/" "-I/home/jerryneedell/Arduino/hardware/Adafruit/samd/cores/arduino" "-I/home/jerryneedell/Arduino/hardware/Adafruit/samd/variants/metro_m4" "/tmp/arduino_build_551525/sketch/Blink.ino.cpp" -o "/dev/null"
Generating function prototypes...
"/home/jerryneedell/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m4 -mthumb -c -g3 -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions  -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10805 -DARDUINO_METRO_M4 -DARDUINO_ARCH_SAMD  -D__SAMD51J19A__ -D__SAMD51__ -DUSB_VID=0x239A -DUSB_PID=0x8020 -DUSBCON '-DUSB_MANUFACTURER="Adafruit LLC"' '-DUSB_PRODUCT="Adafruit Metro M4"' -D__FPU_PRESENT -DARM_MATH_CM4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 "-I/home/jerryneedell/.arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/" "-I/home/jerryneedell/.arduino15/packages/arduino/tools/CMSIS-Atmel/1.1.0/CMSIS/Device/ATMEL/" "-I/home/jerryneedell/Arduino/hardware/Adafruit/samd/cores/arduino" "-I/home/jerryneedell/Arduino/hardware/Adafruit/samd/variants/metro_m4" "/tmp/arduino_build_551525/sketch/Blink.ino.cpp" -o "/tmp/arduino_build_551525/preproc/ctags_target_for_gcc_minus_e.cpp"
"/home/jerryneedell/arduino-1.8.5/tools-builder/ctags/5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "/tmp/arduino_build_551525/preproc/ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"/home/jerryneedell/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m4 -mthumb -c -g3 -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -MMD -DF_CPU=48000000L -DARDUINO=10805 -DARDUINO_METRO_M4 -DARDUINO_ARCH_SAMD  -D__SAMD51J19A__ -D__SAMD51__ -DUSB_VID=0x239A -DUSB_PID=0x8020 -DUSBCON '-DUSB_MANUFACTURER="Adafruit LLC"' '-DUSB_PRODUCT="Adafruit Metro M4"' -D__FPU_PRESENT -DARM_MATH_CM4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 "-I/home/jerryneedell/.arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/" "-I/home/jerryneedell/.arduino15/packages/arduino/tools/CMSIS-Atmel/1.1.0/CMSIS/Device/ATMEL/" "-I/home/jerryneedell/Arduino/hardware/Adafruit/samd/cores/arduino" "-I/home/jerryneedell/Arduino/hardware/Adafruit/samd/variants/metro_m4" "/tmp/arduino_build_551525/sketch/Blink.ino.cpp" -o "/tmp/arduino_build_551525/sketch/Blink.ino.cpp.o"
Compiling libraries...
Compiling core...
"/home/jerryneedell/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m4 -mthumb -c -g3 -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -MMD -DF_CPU=48000000L -DARDUINO=10805 -DARDUINO_METRO_M4 -DARDUINO_ARCH_SAMD  -D__SAMD51J19A__ -D__SAMD51__ -DUSB_VID=0x239A -DUSB_PID=0x8020 -DUSBCON '-DUSB_MANUFACTURER="Adafruit LLC"' '-DUSB_PRODUCT="Adafruit Metro M4"' -D__FPU_PRESENT -DARM_MATH_CM4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 "-I/home/jerryneedell/.arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/" "-I/home/jerryneedell/.arduino15/packages/arduino/tools/CMSIS-Atmel/1.1.0/CMSIS/Device/ATMEL/" "-I/home/jerryneedell/Arduino/hardware/Adafruit/samd/cores/arduino" "-I/home/jerryneedell/Arduino/hardware/Adafruit/samd/variants/metro_m4" "/home/jerryneedell/Arduino/hardware/Adafruit/samd/variants/metro_m4/variant.cpp" -o "/tmp/arduino_build_551525/core/variant.cpp.o"
"/home/jerryneedell/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-gcc" -c -g3 -x assembler-with-cpp -MMD -DF_CPU=48000000L -DARDUINO=10805 -DARDUINO_METRO_M4 -DARDUINO_ARCH_SAMD  -D__SAMD51J19A__ -D__SAMD51__ -DUSB_VID=0x239A -DUSB_PID=0x8020 -DUSBCON '-DUSB_MANUFACTURER="Adafruit LLC"' '-DUSB_PRODUCT="Adafruit Metro M4"' -D__FPU_PRESENT -DARM_MATH_CM4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 "-I/home/jerryneedell/.arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/" "-I/home/jerryneedell/.arduino15/packages/arduino/tools/CMSIS-Atmel/1.1.0/CMSIS/Device/ATMEL/" "-I/home/jerryneedell/Arduino/hardware/Adafruit/samd/cores/arduino" "-I/home/jerryneedell/Arduino/hardware/Adafruit/samd/variants/metro_m4" "/home/jerryneedell/Arduino/hardware/Adafruit/samd/cores/arduino/pulse_asm.S" -o "/tmp/arduino_build_551525/core/pulse_asm.S.o"
"/home/jerryneedell/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-gcc" -mcpu=cortex-m4 -mthumb -c -g3 -Os -w -std=gnu11 -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -MMD -DF_CPU=48000000L -DARDUINO=10805 -DARDUINO_METRO_M4 -DARDUINO_ARCH_SAMD  -D__SAMD51J19A__ -D__SAMD51__ -DUSB_VID=0x239A -DUSB_PID=0x8020 -DUSBCON '-DUSB_MANUFACTURER="Adafruit LLC"' '-DUSB_PRODUCT="Adafruit Metro M4"' -D__FPU_PRESENT -DARM_MATH_CM4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 "-I/home/jerryneedell/.arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/" "-I/home/jerryneedell/.arduino15/packages/arduino/tools/CMSIS-Atmel/1.1.0/CMSIS/Device/ATMEL/" "-I/home/jerryneedell/Arduino/hardware/Adafruit/samd/cores/arduino" "-I/home/jerryneedell/Arduino/hardware/Adafruit/samd/variants/metro_m4" "/home/jerryneedell/Arduino/hardware/Adafruit/samd/cores/ [... cut ...]
An error occurred while uploading the sketch
readWord(addr=0)=0x66616441
readWord(addr=0xe000ed00)=0x74697572
readWord(addr=0x400e0740)=0x72694320
version()=cuitPython 3.0.0-alpha.1-301-g568c04e on 2018-02-27; Metro M4 Express Rev B (Black) with samd51j19
chipId=0x72694320
Unsupported processor
Send auto-baud
Set binary mode
readWord(addr=0)=0x30303077
readWord(addr=0xe000ed00)=0x30303030
readWord(addr=0x400e0740)=0x23342c30
version()=wE000ED00,4#w400E0740,4#V#
chipId=0x23342c30
Unsupported processor
```

tannewt, 2018-03-05:
I have a feeling this is because 3.x doesn't reset into the bootloader after receiving 1200 baud serial.

jepler, 2018-04-04:
False alarm, I don't think #740 is actually related to this.

jerryneedell, 2018-04-25:
I was able to build/upload Arduino Blink with no problem. With Arduino Adafruit SAMD BSP  v1.1.0   and CP3.0-alpha6.  This appears to have been resolved,

closed 2018-04-25 by jerryneedell