adafruit/circuitpython #655: FS does not mount reliably with CP3.0  on itsybitsy_m0_express

jerryneedell, 2018-03-02:
I am finding that when I flash CP 3.0 to my itsybitsy_m0 _express, my Ubuntu linux system is not recognizing the FS. 

It  works fine if I flash CP 2.x

dmesg output when FS fails with 3.0
```

[168355.572364] usb 3-3.2: New USB device found, idVendor=239a, idProduct=000f
[168355.572370] usb 3-3.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[168355.572373] usb 3-3.2: Product: ItsyBitsy M0 Express
[168355.572376] usb 3-3.2: Manufacturer: Adafruit Industries
[168355.573052] cdc_acm 3-3.2:1.0: ttyACM0: USB ACM device
[168355.575705] usb-storage 3-3.2:1.2: USB Mass Storage device detected
[168355.576124] scsi host8: usb-storage 3-3.2:1.2
[168355.577266] hid-generic 0003:239A:000F.0027: hiddev1,hidraw3: USB HID v1.00 Device [Adafruit Industries ItsyBitsy M0 Express] on usb-0000:00:14.0-3.2/input3
[168356.587395] scsi 8:0:0:0: Direct-Access     ItsyBits y M0 Express     1.00 PQ: 0 ANSI: 2
[168356.588349] sd 8:0:0:0: Attached scsi generic sg2 type 0
[168356.589610] sd 8:0:0:0: [sdc] 8000 512-byte logical blocks: (4.10 MB/3.91 MiB)
[168356.589830] sd 8:0:0:0: [sdc] Write Protect is off
[168356.589834] sd 8:0:0:0: [sdc] Mode Sense: 0f 00 00 00
[168356.590016] sd 8:0:0:0: [sdc] No Caching mode page found
[168356.590023] sd 8:0:0:0: [sdc] Assuming drive cache: write through
[168356.598382]  sdc:
[168356.599868] sd 8:0:0:0: [sdc] Attached SCSI removable disk
[168378.003723] usb 3-3.2: USB disconnect, device number 80
[168378.063064] FAT-fs (sdc): unable to read boot sector to mark fs as dirty
```

jerryneedell, 2018-03-02:
This is reproducible - 2.x loads and boots consistently. 3.0 does not . Dotstar turns blue and nothing happens.  I tried connecting it to a Mac with same result.

siddacious, 2018-03-02:
I may have seen some strangeness with 3.0 on the bitsy but I was under the impression that @tannewt recently fixed something with the flash for it that had been causing issues. I'll test again and report back.

tannewt, 2018-03-05:
I am changing it with the DMA work so I'll take a look at it after.

siddacious, 2018-03-11:
Doesn't seem to work for me with the changes you mentioned

tannewt, 2018-03-12:
@siddacious and @jerryneedell can you send me the text from the top of the flash chip? We may have different ones. Feel free to send a picture of the board.

jerryneedell, 2018-03-12:
Sure, but It’ll be late tonight or more likely tomorrow before I can. I have not tried the latest version yet an will do that as well.

tannewt, 2018-03-12:
No rush! Thanks for the help!

tannewt, 2018-03-13:
I think some FS weirdness was introduced with the latest ASF with 9b4477e1dc088d18127838310b828514cb33edea. I'm looking into it now.

jerryneedell, 2018-03-13:
My ItsyBitsy_m0_espress works fine with the latest build:

```
Adafruit CircuitPython 3.0.0-alpha.2-17-g9d928b5-dirty on 2018-03-12; Adafruit Itsy Bitsy M0 Express with samd21g18
>>> 
```
Even the dotstar works!
trying to read the flash chip markings....

jerryneedell, 2018-03-13:
is this the right chip? ￼
[image]

tannewt, 2018-03-13:
Yup! I think so. I'll need to compare with the datasheets tomorrow.

I think it may actually be a code bug too. If you have time, try with commit 9c5a9e34118532bd54d92c9f968534ad9f9409ae (before the latest asf4 update).

jerryneedell, 2018-03-13:
Built BOARD=itsybitsy_m0  (old name) from commit 9c5a9e34118532bd54d92c9f968534ad9f9409ae

This version fails  - no FS mounted - dotstar turns green for a few seconds then turns off - repeats.

jerryneedell, 2018-03-13:
Restored current master - working fine.

tannewt, 2018-03-13:
@jerryneedell thats weird! I expected 9c5a9e3 to work and current master to not work. Going heads down on usb shortly to see what I find.

jerryneedell, 2018-03-13:
This version does not work for me -  itsybitys just goes to a solid blues dotstar - no FS mount
as before:
```
[443568.884692] sd 7:0:0:0: [sdc] Attached SCSI removable disk
[443591.740704] usb 3-3.2: USB disconnect, device number 47
[443591.823432] FAT-fs (sdc): unable to read boot sector to mark fs as dirty
```

jerryneedell, 2018-03-13:
do you want to reopen this one or should I create a new issue.

PR #677 (2018-03-13): Move usb read finish into interrupt.
Having the `active_read = false` in the background function left
a chance that a new_write occurs before active_read is set to false.

In that case, we'll read the appropriate data rather than write it
and never clear the active write.

Hopefully fixes #655.
commit 81572481 (2018-03-13): Move usb read finish into interrupt.

Having the `active_read = false` in the background function left
a chance that a new_write occurs before active_read is set to false.

In that case, we'll read the appropriate data rather than write it
and never clear the active write.

Hopefully fixes #655.
PR #679 (2018-03-14): Check usb_busy up front in usb background function.
Waiting to do so risks accidentally queueing another response.

Hopefully fixes #655 but we'll let @jerryneedell confirm.
commit 7a3f86d1 (2018-03-14): Check usb_busy up front in usb background function.

Waiting to do so risks accidentally queueing another response.

Hopefully fixes #655 but we'll let @jerryneedell confirm.
closed 2018-03-14 by dhalbert