adafruit/circuitpython #660: Safe-mode crash when I make a GamePad with cpx buttons

inventhouse, 2018-03-07:
I tried to use the gamepad module to read the buttons on my CircuitPlayground Express

I added .txt to the filename so it would attach; in case it matters, rather than renaming it, I have a main.py that just imports the script I want to run.

[cpxTempButton.py.txt](https://github.com/adafruit/circuitpython/files/1787969/cpxTempButton.py.txt)

tannewt, 2018-03-07:
This is the same as #568. You need to give it something from `board` rather than from `cpx` because the CPX pin is a DigitalInOut.

deshipu, 2018-03-07:
Actually GamePad expects DigitalInOut objects, so that part of the code is correct.

deshipu, 2018-03-07:
@inventhouse what versionf of CircuitPython is this?

inventhouse, 2018-03-10:
Adafruit CircuitPython 2.2.3 on 2018-02-05; Adafruit CircuitPlayground Express with samd21g18

deshipu, 2018-03-10:
By the way, gamepad will not with CPX buttons, because they are pulled down and when you press them, the connect to VCC, and the gamepad library expects the pins to be pulled up, and when pressed, connected to GND, as is the usual way for handling buttons. I didn't know about CPX buttons when I wrote it.

Nevertheless, it should not crash.

inventhouse, 2018-03-11:
It would be great if gamepad were "ambidextrous", button polling is, as I'm sure you know, a surprisingly tricky business and I don't expect they can/will change the pull-ups on the CPX buttons at this point.

deshipu, 2018-03-11:
Sure, but I'm still thinking about how to do that.

inventhouse, 2018-03-11:
(Though it might be simplest if it's all-or-nothing, "mixed mode" would not only be hairy to implement, it would make for a complex/confusing interface; better to make two GamePad objects, one for "negative-press"/standard buttons and one for positive-press.)

inventhouse, 2018-03-11:
Note: I'm thinking the GamePad init would take a "pressed = True"/"pressed = False" (defaulting to the latter) argument; clients would make two instances if they wanted some buttons pulled-up and some pulled-down.

deshipu, 2018-03-11:
I would rather prefer checking if the DigitalInOut instance has already a pull, and then react accordingly, otherwise default to pull-up.

tannewt, 2018-04-02:
Fixed by #732 and #733. Thanks @deshipu

closed 2018-04-02 by tannewt