adafruit/circuitpython #664: gamepad module crashes cp3.0

deshipu, 2018-03-10:
I didn't report this properly so far, because I wasn't able to pin point exactly what makes it crash. This is what I know:

* When imported and created from the REPL, everything works perfectly fine and there are no problems or crashes.
* When imported and created from a frozen mpy module, it crashes as soon as it tries to read the pin states inside the tick. The crash completely disables USB communication, so the disk becomes unresponsive too.
* The crash happens in the tick function when it tries to read the digital pin state. Commenting out that line makes it not crash (but of course it doesn't read the keys then).
* Creating an empty gamepad with no pins to check also makes it not crash.
* Various combinations of mpy-compiling, freezing, just importing, importing from repl, importing from main.py, etc. make it crash or not — I didn't test it enough to put my finger on when it works exactly and when it doesn't.

tannewt, 2018-03-10:
Did you remember to uncomment gamepad_reset in port.c?

deshipu, 2018-03-10:
Yes. I also found and fixed a potential race condition in initialization, but that didn't help.

deshipu, 2018-03-11:
I just tested this today on the build-metro_m4_express_revb and it seems to work. The cases I tested:

* imported and initialized from REPL
* imported and initialized in main.py
* imported and initialized in a .py module imported from main.py
* imported and initialized in a .mpy module imported from main.py
* imported and initialized in a frozen mpy module imported from main.py

I will test on the M0 to make sure, but maybe the bug got magically fixed somehow.

deshipu, 2018-03-11:
I tested it on feather_m0_express and the bug seems to be gone.

closed 2018-03-11 by deshipu