adafruit/circuitpython #671: Implement time.localtime()

notro, 2018-03-11:
I want to emulate a DS1307 on a Feather M0 so I need access to clock time. Has anyone given this some thought?
What the internal representation should be: seconds or struct tm, and how should the clock be set, board.setclock()?
I see that the samd21 clock can run in several modes.

Code so far:
```c
inline uint64_t common_hal_time_time() {
    return (18 * 365 + 74) * 86400 + (19 * 60 + 4) * 60;
}

mp_obj_t struct_time_new(size_t n, const mp_obj_t *items) {

    // TODO: How to construct a struct_time?

    return MP_OBJ_FROM_PTR(mp_obj_new_tuple(n, items));
};

//| .. method:: time()
//|
//|   Return the current time in seconds since the Epoch.
//|   Fractions of a second may be present if the system clock provides them.
//|
//|   :return: the current time
//|   :rtype: int
//|
STATIC mp_obj_t time_time(void) {
    return mp_obj_new_int(common_hal_time_time());
}
MP_DEFINE_CONST_FUN_OBJ_0(time_time_obj, time_time);

//| .. method:: localtime([secs])
//|
//|   Convert a time expressed in seconds since Jan 1, 2000 to a struct_time in
//|   local time. If secs is not provided or None, the current time as returned
//|   by time() is used.
STATIC mp_obj_t time_localtime(size_t n_args, const mp_obj_t *args) {
    timeutils_struct_time_t tm;
    mp_obj_t elems[9];
    mp_int_t secs;

    if (n_args == 0 || args[0] == mp_const_none) {
        secs = common_hal_time_time();
    } else {
        secs = mp_obj_get_int(args[0]);
    }

    timeutils_seconds_since_2000_to_struct_time(secs, &tm);

    elems[0] = mp_obj_new_int(tm.tm_year);
    elems[1] = mp_obj_new_int(tm.tm_mon);
    elems[2] = mp_obj_new_int(tm.tm_mday);
    elems[3] = mp_obj_new_int(tm.tm_hour);
    elems[4] = mp_obj_new_int(tm.tm_min);
    elems[5] = mp_obj_new_int(tm.tm_sec);
    elems[6] = mp_obj_new_int(tm.tm_wday);
    elems[7] = mp_obj_new_int(tm.tm_yday);
    elems[8] = mp_obj_new_int(-1);

    return struct_time_new(9, elems);
}
MP_DEFINE_CONST_FUN_OBJ_VAR_BETWEEN(time_localtime_obj, 0, 1, time_localtime);

//| .. method:: mktime(t)
//|
//|   This is the inverse function of localtime(). Its argument is the
//|   struct_time or full 9-tuple (since the dst flag is needed; use -1 as the
//|   dst flag if it is unknown) which expresses the time in local time, not UTC.
//|   The earliest date for which it can generate a time is Jan 1, 2000.
STATIC mp_obj_t time_mktime(mp_obj_t t) {
    mp_obj_t *elem;
    size_t len;

    if (!MP_OBJ_IS_TYPE(t, &mp_type_tuple) && !MP_OBJ_IS_TYPE(t, MP_OBJ_FROM_PTR(&struct_time_type_obj))) {
        mp_raise_TypeError("Tuple or struct_time argument required");
    }

    mp_obj_tuple_get(t, &len, &elem);
    if (len != 9) {
        mp_raise_TypeError("function takes exactly 9 arguments");
    }

    return mp_obj_new_int_from_uint(timeutils_mktime(mp_obj_get_int(elem[0]), mp_obj_get_int(elem[1]), mp_obj_get_int(elem[2]),
                                                     mp_obj_get_int(elem[3]), mp_obj_get_int(elem[4]), mp_obj_get_int(elem[5])));
}
MP_DEFINE_CONST_FUN_OBJ_1(time_mktime_obj, time_mktime);

```
(full [diff](https://gist.github.com/notro/6fa81d3e948aefd3053cd2b1df8ead41))

tannewt, 2018-03-12:
I'd probably add an `rtc` module which has a `native` singleton copy of an `RTC` class. Then the RTC class could be used just like the DS1307. The `rtc` module itself could allow you to set a default RTC to be used by the `time` module. That way you could use a Python-implemented RTC potentially.

notro, 2018-03-18:
This is as far as I got this weekend following your suggestion ([diff](https://gist.github.com/notro/8cf8d57fd99e88e7f7ee5bd4f5ecff51)):
```
>>> import rtc
>>> import time
>>> rtc.RTC() == rtc.RTC()
True
>>> rtc.RTC().datetime()
(2018, 3, 17, 23, 58, 10, 0, 0, -1)
>>> time.localtime()
(2018, 3, 17, 23, 58, 10, 0, 0, -1)
>>>
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== class RTC(object):
===     def datetime(self):
===         return (2018, 3, 17, 21, 1, 47, 5, 76, 0)
===
=== r = RTC()
=== rtc.set_time_source(r)
===
>>> time.localtime()
(2018, 3, 17, 21, 1, 47, 5, 76, 0)
>>> 
```

I could use some help in creating a struct_time from C, currently I'm cheating by using a tuple:
```c
mp_obj_t struct_time_from_tm(timeutils_struct_time_t *tm) {
    mp_obj_t elems[9];

    elems[0] = mp_obj_new_int(tm->tm_year);
    elems[1] = mp_obj_new_int(tm->tm_mon);
    elems[2] = mp_obj_new_int(tm->tm_mday);
    elems[3] = mp_obj_new_int(tm->tm_hour);
    elems[4] = mp_obj_new_int(tm->tm_min);
    elems[5] = mp_obj_new_int(tm->tm_sec);
    elems[6] = mp_obj_new_int(tm->tm_wday);
    elems[7] = mp_obj_new_int(tm->tm_yday);
    elems[8] = mp_obj_new_int(-1);

    // TODO: How to construct a struct_time?

    return MP_OBJ_FROM_PTR(mp_obj_new_tuple(9, elems));
};

```
The next step is to hook it up to the hw rtc.

tannewt, 2018-03-19:
Good job so far @notro! I think this will help you create a struct_time: https://github.com/adafruit/circuitpython/blob/master/shared-bindings/time/__init__.c#L82

Its actually just a named tuple. I believe the type you'll want to pass is `&struct_time_type_obj`.

notro, 2018-03-25:
Yay, I have a running clock!
```
>>> time.localtime()
struct_time(tm_year=2018, tm_mon=3, tm_mday=25, tm_hour=23, tm_min=36, tm_sec=25, tm_wday=6, tm_yday=84, tm_isdst=-1)
```

[rtc module diff](https://gist.github.com/notro/72ad0fd4bd2934cad43e87729e65fe1a)
[asf4 diff](https://gist.github.com/notro/e087c42e5cc2c79f81ba85faccfd9b8d)
[samd21 diff](https://gist.github.com/notro/2e9eca8cbb9e7f12ec416043eb0fd524)

I'll see if I can make a PR next weekend so we can discuss the details.

tannewt, 2018-03-26:
Awesome work! A early PR would be very welcome!

If you are only able to be online weekends we can coordinate times so we can chat. I usually try to avoid work on the weekend but can trade weekday time to weekend time to coordinate with folks. Let me know. ([Discord](https://adafru.it/discord) is an easy way to contact me too.)

notro, 2018-03-26:
Thanks for the offer, but I'm not much of a chatter :-)

I mostly do Linux kernel development so I'm used to discussions on mail which suits me fine.

And since I love reading code more than writing it, I prefer to dig around the code base to answer my own questions when I'm stuck. I've set up [elixir](https://github.com/free-electrons/elixir) so I can browse the CP source like I can do with Linux: https://elixir.bootlin.com/linux/latest/ident/start_kernel

Ofc the code itself isn't so good at answering the cultural aspects of a project, so I rely on the review process for that. Since this is my first contribution, I expect it to be a few rounds before you're happy :-)

Let me know if this isn't how things are done in this project.

tannewt, 2018-03-26:
@notro All good! Feel free to email me at scott@adafruit.com if you have questions. I'm happy to talk about cultural stuff.

closed 2018-04-07 by notro