adafruit/circuitpython #672: USB Midi!

tannewt, 2018-03-12:
Requested by JohnPark and Noe. Then we can create simple button instruments using midi.

djecken, 2018-03-12:
This is a great "example" project for USB MIDI. Simple button presses for triggering midi notes. https://learn.adafruit.com/arcade-button-control-box

moliata, 2018-12-12:
Any update on this?

ladyada, 2018-12-12:
moli, we had to refactor all of the usb stack but we’re getting there, subscribe to this topic and youll kniw the moment its available. no ETA 😁

moliata, 2018-12-13:
@ladyada Awesome, thanks!

moliata, 2018-12-16:
@ladyada following up, even tho' there is no ETA, do you know is it a couple of days or a few months job? Thanks!

ladyada, 2018-12-16:
no ETA 😁

closed 2019-01-11 by dhalbert

kevinjwalters, 2020-03-15:
For anyone who was following this ticket which was mentioned in https://forums.adafruit.com/viewtopic.php?f=58&t=141504, the CircuitPython library has been around for a while now and can be found in the standard bundle and on GitHub as https://github.com/adafruit/Adafruit_CircuitPython_MIDI

PR #1449 (2019-01-11): Add USB MIDI support for SAMD and nRF.
The API should be identical to using a UART for MIDI.

Fixes #672
commit b5e40f52 (2019-01-11): Add USB MIDI support for SAMD and nRF.

The API should be identical to using a UART for MIDI.

Fixes #672