adafruit/circuitpython #673: DMX (lighting protocol)

tannewt, 2018-03-12:
This is used by DJs to control lighting.

siddacious, 2018-03-12:
If this ever gets done a great project would be an Artemis "red alert" siren:

http://artemiswiki.pbworks.com/w/page/68387358/DMX%20Interface

siddacious, 2018-10-03:
I found these which might be useful reference
https://github.com/jaydcarlson/microcontroller-test-code/tree/master/SAMD10/dmx
https://github.com/claudeheintz/LXSAMD21DMX

s-light, 2018-11-28:
some more references i found on my search that could be helpfull:
- [Generating DMX512 - simple demo](https://forum.micropython.org/viewtopic.php?t=2164) pointing to [clacktronics/pyb_dmx](https://github.com/clacktronics/pyb_dmx)
- [claudeheintz/LXESP32DMX](https://github.com/claudeheintz/LXESP32DMX)
- [chrisstaite/TeensyDmx](https://github.com/chrisstaite/TeensyDmx) (active development)

ariherzig, 2020-01-07:
I'm also very interested in there being a way to send and receive DMX-512 data with CircuitPython.

tannewt, 2020-01-07:
Can this be done with UART at the lowest level? Do we just need a Python library for it?

What is the least expensive testing setup?

wallarug, 2020-01-08:
Hey @tannewt ,

A Python library would be the easiest way to do it.  But it might need doing at the hardware level of the SAMD21/51.   I read in the datasheet that these chips should support RS485 at the hardware level.

I know only one python library that does this well at the moment.  https://github.com/mathertel/DMXSerial

At the heart of DMX:
* serial RS485
* 250 kbit/s
* 513 byte transactions (frames)

[Element 14 DMX Explained](https://www.element14.com/community/groups/open-source-hardware/blog/2017/08/24/dmx-explained-dmx512-and-rs-485-protocol-detail-for-lighting-applications)

wallarug, 2020-01-08:
After extensive reading from the [adafruit forums from 2018](https://forums.adafruit.com/viewtopic.php?f=8&t=142073&start=45) ... please ignore my comments about direct hardware support.

The SAMD51 has some support for RS485 but external hardware is still required.  The SAMD21 has no built in support.

The only way is to use a transceiver module (MAX3485 is most popular) and normal UART protocols with a packet decoding layer (presumably this is what everyone in this issue is looking for).

I have been looking into this for about 7 years and been doing stage lighting for 10+ years - that's why I'm interested 😄 .

s-light, 2020-01-08:
@tannewt least expensive could be some 'amazon dmx led spot' (you never know if they are conform to the specifications...) -
the better setup: a scope or digital-logger/analyzer thing... ;-)
or just some already 'proven to work' arduino library..
all the implementation can be tested in the 3.3V / 5V TTL levels...
as @wallarug  has written - to get to the actuall DMX512 HW-Layer ('just' RS485) some differential driver /receiver is needed.. 
here would be the challenge to check and find one that is 3.3V compatible..
out of my head i don't know one.. (but there are lots of options form different manufactures - [example ti](http://www.ti.com/interface/rs-485-rs-422/products.html?pqs=paqs&familyid=545#p305nom=3.3;4&p2882=1;9&p2881=1;1&p976max=0.25;1&sort=p1130;asc))

theoretically you could do it with just an UART implementation..
but this could get hacky - as you need to fastly switch baudrates to generate the stop bit timming.. (if i remember correctly)
lower level - maybe the 'USART' support in the chip can handle this thing already in a smoother way...

@wallarug your example is only partly correct:
DMX is not specified to transmit all 512 channels. you are allowed to send less!
and sometimes that is usefull! in my classic arduino projects i used this to reduce performance /memory problems ;-) (only sending the ~12 channels the device needed...)

tannewt, 2020-01-08:
Thanks for the info. To summarize: I don't think there is any core work needed for this. On the drivers side we'd likely need a community library for the external transmitter IC and then a library for the DMX protocol on top of that.

I don't have any experience with DMX so someone else will be a better person to do it. I'm happy to consult on the CircuitPython side of things but won't be picking this up directly myself.

s-light, 2020-01-08:
the external IC is only 'levelshifting' so no lib there..
maybe we need some special edge cases implemented in the UART Core...
in total you are right - best is if someone with deep detail understanding of the DMX-Protocoll finds the time to implement it -and then we will know where the limits are..
i think i could do it - but i don't know if and when i find spare time for it - so if there is someone else willing i could support/help with some protocoll information and testing with some hardware-setups.

Tasm-Devil, 2020-04-20:
U can start with this fragment:

`
import board
import busio
import digitalio
import time
from array import array
import gc

class DMX():
    def __init__(self, max_channels):
        # First byte is always 0, 512 after that is the 512 channels
        self.dmx_message = array('B', [0] * (max_channels+1))
        self.dmx_uart = digitalio.DigitalInOut(board.TX)
        self.dmx_uart.direction = digitalio.Direction.OUTPUT
        self.dmx_uart.value = 1

    def set_channels(self, message):
        """
        a dict and writes them to the array
        format {channel:value}
        """
        for ch in message:
            self.dmx_message[ch] = message[ch]

    def write_frame(self):
        """
        Send a DMX frame
        """
        self.dmx_uart.value = 0
        time.sleep(88E-6)
        self.dmx_uart.value = 1
        self.dmx_uart.deinit()
        time.sleep(8E-6)
        self.dmx_uart = busio.UART(tx=board.TX,rx=None, baudrate=250000, bits=8, parity=None, stop=2)
        self.dmx_uart.write(self.dmx_message)
        self.dmx_uart.deinit()
        self.dmx_uart = digitalio.DigitalInOut(board.TX)
        self.dmx_uart.direction = digitalio.Direction.OUTPUT
        self.dmx_uart.value = 1

channel = 10

dmx = DMX(channel)
#dmx.set_channels({1:0})
for i in range(1,channel):
    print(i)
    dmx.set_channels({i:255})
    dmx.write_frame()
    time.sleep(2)
`

ariherzig, 2020-04-21:
Thank you for this fragment, it is super helpful for me to try and wrap my head around how to get DMX working with Circuitpython!

DatanoiseTV, 2023-07-13:
It has been stale for over 2y. Is there any news on this?

s-light, 2023-07-13:
currently i have no use-cases for dmx...
otherwise i would try the fragment from @Tasm-Devil.
seems like a valid first starting point...

wallarug, 2023-07-13:
> It has been stale for over 2y. Is there any news on this?

Just to confirm, is all that people want is a CircuitPython Library that does DMX control?  Nothing more that this?

s-light, 2023-07-14:
> > It has been stale for over 2y. Is there any news on this?
> 
> Just to confirm, is all that people want is a CircuitPython Library that does DMX control? Nothing more that this?

if i think about it for a moment - 
i can imagine three simple scenarios:
1. DMX receiving - so the CP device can act as device in an event/stage enviroment
1. DMX sending - simple just some channels in a buffer that get send manually or automatically every x ms
1. DMX sending - API compatible to use it with [FancyLED](https://learn.adafruit.com/fancyled-library-for-circuitpython/overview) - maybe with a additionall mapping layer - so we could use this to integrate dmx-led-spots or similar things in an FancyLED Project....

as told - just a quick brainstorming...
and i think the *sketch* from Tasm-Devil  is  a good starting point for 2.

mydana, 2023-07-27:
I’m working on two libraries, one for DMX sending, and the other for DMX receiving. Both use RP2040’s PIO. They both work. I’m now doing some clean-up and hope I can get them into the Community Bundle.

wallarug, 2023-07-27:
> I’m working on two libraries, one for DMX sending, and the other for DMX receiving. Both use RP2040’s PIO. They both work. I’m now doing some clean-up and hope I can get them into the Community Bundle.

Awesome!  Can you provide some GitHub links when you are ready?

dhalbert, 2023-07-28:
> I’m working on two libraries, one for DMX sending, and the other for DMX receiving. Both use RP2040’s PIO. They both work. I’m now doing some clean-up and hope I can get them into the Community Bundle.

We are really pleased to hear this!

mydana, 2023-09-18:
https://github.com/mydana/CircuitPython_DMX_Transmitter Would love feedback.

wallarug, 2023-09-18:
> https://github.com/mydana/CircuitPython_DMX_Transmitter Would love feedback.

That looks awesome. I'll take another look soon and comment in the issues,

kylefmohr, 2024-02-24:
> https://github.com/mydana/CircuitPython_DMX_Transmitter Would love feedback.

Just wanted to say thanks @mydana, I just found your library from this thread and was able to set it up and get it working without any issues using an RP2040 and CircuitPython 9.0.0 beta 2. Your wiring diagram in the README was also very helpful. Thank you! I hope that it can be added to the community bundle soon.

mydana, 2024-05-18:
https://github.com/mydana/CircuitPython_DMX_Receiver

It's still a bit buggy, but I'd love feedback.

tschundler, 2024-08-14:
I independently came up with something very similar to https://github.com/adafruit/circuitpython/issues/673#issuecomment-616831705 from @Tasm-Devil switching the transmit pin between digital out (for Break + Mark-after-Break) and uart (for start condition + data)

That works on the ESP32s3 I use.

Notable difference is that `time.sleep` has a minimum increment of 1ms. to asking it to sleep 8uS for the Mark-after-break ends up being really 1ms. But initializing the uart takes more than the minimum MAB time, so I just set the pin high and don't sleep.

---

I also experimented with using SPI to send out the bitstream. That allows very precise time control, but didn't really make any difference in my testing.

maxaitel, 2024-09-13:
> https://github.com/mydana/CircuitPython_DMX_Transmitter Would love feedback.

Why won't circup see your package?

```dmx_transmitter is not a known CircuitPython library.```

:(

I'm on a pico plus 2 running CircuitPython 9.

tannewt, 2024-09-16:
> Why won't circup see your package? @mydana

I needs to be added to the community bundle for circup to see it: https://github.com/adafruit/CircuitPython_Community_Bundle/tree/main/libraries/helpers

mydana, 2024-10-16:
@tannewt I plan to simplify and add a new version to the community bundle. I'm using the library myself to drive some DMX strip lights, and I have some lessons learned that I will incorporate, but will change the interface a bit thereby making it more compatible with the Adafruit animations. 

@maxaitel if you haven't done it already, please create a folder called dmx_transmitter into your microcontoller's lib folder, and copy all the .py files from dmx_transmitter folder in the GitHub repository you'd linked above. Please let me know if it works for you.

tschundler, 2024-10-17:
Is RP2040-only library suitable for for bundle? Or do you have a fall-back like @Tasm-Devil's for others?

Some of my code:
```
import busio
import time
import digitalio

def delay_us(us):
    w = time.monotonic_ns() + us * 1000
    while time.monotonic_ns() < w:
        pass

class DMXSerial:
    def __init__(self, txpin):
        self._start = bytearray(1)
        self._txpin = txpin
    def send(self, data):
        with digitalio.DigitalInOut(self._txpin) as dio:
            dio.switch_to_output(value=True)
            dio.value = False # break
            delay_us(100) # 100us break
            # 12us Mark after break (high)
            dio.value = True # Mark after break
            # switch to UART is slow
            #delay_us(12) # 12us MAB
        with busio.UART(tx=self._txpin, rx=None, baudrate=250000,bits=8, parity=None, stop=2) as dmx_uart:
            dmx_uart.write(self._start)
            dmx_uart.write(data)
```

(I'm sending it a bytearray as the data)

tannewt, 2024-10-17:
Thank you @mydana!

> Is RP2040-only library suitable for for bundle?

RP2040-only is totally fine for the bundle. It'd be helpful to note the restriction in the README for it.