adafruit/circuitpython #674: OSError: 19 when attempting storage.mount(vfs, "/sd") for Adalogger Featherwing

waldo323, 2018-03-13:
using an feather M0 with adalogger featherwing in trippler I get an OSError: 19 when trying to run storage.mount(vfs, "/sd")

REPL:
>>> import board
>>> import busio
>>> import digitalio
>>> import adafruit_sdcard
>>> import storage
>>>
>>> spi = busio.SPI(board.SCK, MOSI=board.MOSI, MISO=board.MISO)
>>> # Use board.SD_CS for Feather M0 Adalogger
>>> cs = digitalio.DigitalInOut(board.**D10**)
>>>
>>> sdcard = adafruit_sdcard.SDCard(spi, cs)
>>> vfs = storage.VfsFat(sdcard)
>>>
>>> storage.mount(vfs, "/sd")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: 19

I also got this error when running one of the examples (modified with D10 for CS)

additional information:
looking at the guide: the https://learn.adafruit.com/adafruit-adalogger-featherwing/circuitpython   
I got to the bit about cs = digitalio...it states this line is for feather m0 so I think this is where things differ and I am second guessing whether this is the right guide. When I do a dir(board) I don't see SD_CS

tannewt - Today at 7:11 PM
looks like board.D10 would work
SD_CS is for the feather adalogger (not wing)

waldo323, 2018-03-14:
update from yesterday issue I was having using feather m0 express with adalogger featherwing: I switched from a 16 GB, to 8GB didn't change code or version of circuit python yet but it 'just worked' got a hello world to print to a file on the 8gb board
the guide at: https://learn.adafruit.com/adafruit-adalogger-featherwing/using-the-sd-card 
states:
"The FeatherWing kit doesn't come with an SD card but we carry one in the shop that is guaranteed to work. Pretty much any SD card should work but be aware that some cheap cards are 'fakes' and can cause headaches." 
I believe size not brand is possibly the issue here but this is just my speculation

waldo323, 2018-03-14:
the main issue is the guide says one thing and board operates differently

possible solutions:
- update library or core code to work with larger cards
- update the guide / documentation to reflect the size should be within certain working values.

siddacious, 2018-03-14:
Is it possible that one card was formatted differently from the other?

waldo323, 2018-03-14:
both are SanDisk
both are formatted fat32 with gparted on an ubuntu 16.04 laptop

in attached picture:

- 16 gb on the left did not work,  
- 8 gb on the left did work.

[image]

waldo323, 2018-03-14:
I did notice there was a 1 meg spot non allocated space before the partition for the 8 gb card and no non allocated space on the 16 gb card

dhalbert, 2018-03-14:
Could you completely erase the 8GB card and reformat it so that it doesn't have an odd partition layout and then try again?

ladyada, 2018-03-14:
use SD formatter if you can - https://www.sdcard.org/downloads/formatter_4/
its the 'official' method of formatting an SD card to FAT filesys

waldo323, 2018-03-14:
Oops, my image caption included two lefts. (That's just not right)
I'm glad the cards are labeled. 

Is there an official way to format the SD cards for Linux?

In the meantime, I will format both cards with my windows laptop and the official method when I get home from work, then try both sd cards with the Featherwing Adalogger. 

Thanks :)

dhalbert, 2018-03-14:
For Linux, you could use `gparted`, `fdisk`, `gnome-disk-utility` (aka "Disks" in a Gnome-ish UI). You can delete any existing partitions and then reformat.

waldo323, 2018-03-15:
Using Windows version of the SD Card Formatter with the 16 GB card met with success!
the 8 GB formatted with the SD Card Formatter continued to meet with success. (maybe they are friends?)

after formatting in windows, testing on the adalogger, I checked with gparted to see what the partitions look like (they both look different from before) they both have 4 MB not allocated at the beginning of the cards.
[image]
[image]

tannewt, 2018-03-22:
Is this still an issue?

kattni, 2018-05-02:
Closing issue. If you run into problems again, please feel free to reopen the issue!

closed 2018-05-02 by kattni