adafruit/circuitpython #680: dotstar not accessible on Gemma_m0 - with CP3.0

jerryneedell, 2018-03-15:
Hmmm - tried dotstart test on gemma_m0  - failing like the trinket_m0 used to....

```
Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 3.0.0-alpha.3-dirty on 2018-03-14; Adafruit Gemma M0 with samd21e18
>>> 
>>> 
>>> 
>>> from board import *
>>> import time
>>> import adafruit_dotstar as dotstar
>>> pixel = dotstar.DotStar(APA102_SCK, APA102_MOSI, 1, brightness=0.1)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "adafruit_dotstar.py", line 74, in __init__
ValueError: Pin PA00 in use
>>> 
```

jerryneedell, 2018-03-15:
looks a lot like #514

jerryneedell, 2018-03-15:
same issue as #514 - uncommenting https://github.com/adafruit/circuitpython/blob/master/ports/atmel-samd/boards/gemma_m0/mpconfigboard.h#L4  fixes it. PR comming

tannewt, 2018-03-15:
Fixed by 04b4026666d8a8ef79a1f2ba14330863424936fc

closed 2018-03-15 by tannewt