adafruit/circuitpython #682: Documentation link does not work

mlmcnees, 2018-03-15:
I've not been using CircuitPython for long but the last several release pages that I have looked through I click the documentation link and the link from GitHub release page that appears as:

Documentation is available in readthedocs.io.

has a link:

http://circuitpython.readthedocs.io/en/latest/index.html

that does not work it should probably be http://circuitpython.readthedocs.io/en/latest/

willingc, 2018-03-17:
Hi @mlmcnees, Thanks for the report, and I'm sorry some of the links sent you to the ReadtheDocs maze page. The team has been actively improving and refactoring docs.

You may already be aware, but I'll post here for others who may view this issue at a later date:
- Docs can be found at https://circuitpython.readthedocs.io and that will direct you to the version specified as the default.
- Other versions of the documentation can be accessed from the bottom of the sidebar navigation triangle for the version dropdown:
[image]

- Due to refactoring some of the links from the GitHub release page and triangle dropdown may not redirect to the documentation for the particular version. The "latest" version will have the most up-to-date docs.

Team, 

There are a few places to look to sync up the versions moving forward:
- check `conf.py` and set the master toctree page [[ref from jupyter_client docs]](https://github.com/jupyter/jupyter_client/blob/master/docs/conf.py#L52)
- during heavy refactoring consider limiting the docs that are displayed to latest, stable, and major releases (set in the RTD interface)
- consider setting a `__version__` and use that in all the places that require version [[ref: `conf.py`]](https://github.com/jupyter/jupyter_client/blob/master/docs/conf.py#L61) [[ref: `_version.py`]](https://github.com/jupyter/jupyter_client/blob/master/jupyter_client/_version.py) [[ref: `__init__.py`]](https://github.com/jupyter/jupyter_client/blob/master/jupyter_client/__init__.py) [[ref: `setup.py`]](https://github.com/jupyter/jupyter_client/blob/master/setup.py)

kattni, 2018-03-17:
This was an excellent way to let us know! If possible, can you post a link to a couple of the pages where you encountered the incorrect link? As @willingc stated, we're currently updating our documentation, and I want to make sure we're catching everything.

sommersoft, 2018-03-17:
Just for clarification, this issue was with the CircuitPython Release Notes, for each release. That being said, @willingc excellent recommendations (as always)! And, thank you from me as well @mlmcnees for letting us know.

Here is a screenshot showing the discrepancy:
[image]

kattni, 2018-03-17:
Thanks to @sommersoft for clarifying where the issue was. I have updated the release notes for 3.0.0a3 and 2.2.4 to reflect the correct Read the Docs links. Thank you again to @mlmcnees for catching this issue! It should be resolved going forward.

closed 2018-03-17 by kattni