adafruit/circuitpython #684: Disable dot release docs in RTD that indirect incorrectly

willingc, 2018-03-18:
This is a minor housekeeping follow up to #682. @sommersoft @kattni @tannewt 

### RTD Versions page
[image]

### Landing page for 2.2.1, 2.2.0, and 2.1.0
[image]

### RTD Admin -> Versions page (using JupyterHub as an example)

As @kattni has updated the links in the GitHub release notes to redirect to 2.x :sunny:, I would recommend that the docs for 2.2.1, 2.2.0, and 2.1.0 be disabled (uncheck `Active`). At present, the RTD UI doesn't offer a good way to change the SHA-1 commit chosen for a release after the release has happened.

[image]

kattni, 2018-03-18:
This is an excellent suggestion. @willingc, thank you for outlining the process! I've disabled 2.2.1, 2.2.0, and 2.1.0 in Read the Docs, leaving latest and 2.x as the active options. So, we should be set for now. Thanks again!

closed 2018-03-18 by kattni

willingc, 2018-03-19:
Thanks @kattni. Happy to share the process. It took me a long time to find out some of the RTD workflow stuff like this.