adafruit/circuitpython #689: Auto add .metadata_never_index to new filesystems

tannewt, 2018-03-21:
This file prevents Mac OSx from cluttering the filesystem with index files.

jepler, 2018-03-22:
I wonder how this will interact with the process done at adafruit that loads the initial content onto the flash filesystem; is putting this circuitpython the right thing?

I have a compile-tested patch, but haven't run it on a device.
~~~~
diff --git a/ports/atmel-samd/supervisor/filesystem.c b/ports/atmel-samd/supervisor/filesystem.c
index 45dc5667b..8819e45cb 100644
--- a/ports/atmel-samd/supervisor/filesystem.c
+++ b/ports/atmel-samd/supervisor/filesystem.c
@@ -66,6 +66,11 @@ void filesystem_init(bool create_allowed) {
         // set label
         f_setlabel(&vfs_fat->fatfs, "CIRCUITPY");
         flash_flush();
+
+        // inhibit file indexing on MacOS
+        FIL fp;
+        f_open(&vfs_fat->fatfs, &fp, "/.metadata_never_index", FA_WRITE | FA_CREATE_ALWAYS);
+        f_close(&fp);
     } else if (res != FR_OK) {
         return;
     }
~~~~

jerryneedell, 2018-03-22:
Since putting the initial content on the board requires that the FS be present, this section of the code should not get run after the demo scripts are loaded. It will only get run if the FS is being recreated. If the demo code were there, it would get wiped anyway.

Just a few thoughts as to adding this to all builds. I am not convinced it is the right thing to do. MacCOS users are a subset of the users. On the smaller system, Trinket_m0, Gemma_m0  will this take a "node" out of service resulting in a a loss of 512 bytes from the FS for all users. I think it better for MacOS users to place this file as part of their initial setup if desired. It can be added to the Learn Guide. Would it be sufficient to have MacOS users execute:
```
touch /media/<USERNAME>/CIRCUITPY/.metadata_never_index
```

jerryneedell, 2018-03-22:
If this is implemented, would it also be necessary to implement it in the /lib folder as well?

jepler, 2018-03-22:
I think my patch above is probably missing a call to flash_flush(), and/o the old call should be moved below the file creation.

I don't actually have a MacOS machine to test, I just picked up this issue because it "looked easy" and was tagged accordingly :wink:

According to 'df' running on my Linux machine, creating a zero-byte file in the root of the filesystem doesn't consume any blocks of storage.  However, it does take one away from the total number of files that can be created; the limit seems to be 128 files+directories.

jepler, 2018-03-22:
.. I noticed this in the README of the trinket m0's default contents from adafruit:
~~~~
The Trinket has a very tiny disk drive so we have disabled Mac OS X indexing
which could take up that valuable space.

So *please* do not remove the empty .fseventsd/no_log, .metadata_never_index
or .Trashes files!
~~~~
so at adafruit they do think this is a useful step to take, for whatever that's worth.

jerryneedell, 2018-03-22:
Yes, they did this for the Trinket M0 when the files are loaded. It is not "built into" CP for the trinket M0.  My point was that if something is going to be built in, does all of the above need to be added in order to be effective.  ( .fseventsd/no_log, .metadata_never_index
or .Trashes files )

jerryneedell, 2018-03-22:
How bad is the impact if these are not added? I don't often use MacOS for accessing these boards.

I do have a MacOS system and a Linux system so I can test the impacts on those, but I don't have a Windows system. Does any of this cause any issues for Windows users and are there any similar Windows file system quirks to me covered as well.

tannewt, 2018-03-22:
I think this is worth it because non-Mac users can always delete the file to recover the entries. Its way more difficult for a Mac user to delete all the extra files made the very first time its plugged in.

@jepler I don't think you need to actually test with Mac. I use it, added that file and it prevented the indexing.

When I'm testing this code I usually force the if statement to be true. (Back up your files first!)

PR #691 (2018-03-22): Autocreate files that prevent MacOS indexing of the CIRCUITPYTHON dive
.. the price of this appears to be about 112 bytes of flash and 12
bytes of RAM, according to the stats printed during the build.

It also uses up 4 directory entries (out of 128), but does not reduce
the number of blocks usable for storing file contents.

These are the same items noted in the Adafruit README for Trinket M0
as preventing MacOS indexing.

Closes: #689
commit a7e3c74f (2018-03-22): Autocreate files that prevent MacOS indexing of the CIRCUITPYTHON dive

.. the price of this appears to be about 112 bytes of flash and 12
bytes of RAM, according to the stats printed during the build.

It also uses up 4 directory entries (out of 128), but does not reduce
the number of blocks usable for storing file contents.

These are the same items noted in the Adafruit README for Trinket M0
as preventing MacOS indexing.

Closes: #689
closed 2018-03-22 by tannewt