adafruit/circuitpython #703: Ethernet FeatherWing Library

sommersoft, 2018-03-24:
A user in Discord was looking for an Ethernet FeatherWing library. After discovering there isn't one yet, they mentioned that they were going to port over the Arduino library. I've given them the guide to build a standard library, and mentioned that you all will probably want it in the Adafruit Bundle.

Can someone create the repo for the library for them?

sommersoft, 2018-03-24:
They didn't mention their targeted platform, but I passed along the issue and question. Will let them respond if they desire...

ladyada, 2018-03-24:
all good, i *was* thinking last night about how we could go about adding support, but i would def do so on an M4 :D

jakebird451, 2018-03-24:
Hello, this is UnknownError from discord. My initial target would be the Adafruit WIZ5500 wiznet ethernet featherwing purely out of necessity. I also have the Adafruit Feather M0 WiFi w/ATWINC1500, but I will have to work on that one later.

ladyada, 2018-03-24:
ooh a pure python implementation definitely wont fit on an m0, especially without external flash. you *might* be able to fit into C - it looks like micropython (which we derive from) has a wiznet5k module!
https://forum.micropython.org/viewtopic.php?t=378
we haven't tried it at all, but your best bet is to use a feather M0 express so you get that extra flash space, then try to compile circuitpython with the wiznet module enabled :) alternately, you could try waiting for the M4 boards (tons more space) or use a pyboard (which supports micropython and thus could use the wiznet module)
note the wiznet module seems to be for the earlier 5200 not the 5500 
https://forum.micropython.org/viewtopic.php?t=2515
so altogether, its a bit of work, but its at least started for you! :)

ladyada, 2018-03-24:
chatted with @tannewt - we may try compiling this into a future circuitpython build from upstream!

tannewt, 2018-03-26:
Yup! I'd be great to have this.

@jakebird451 what do you want to do with it? HTTP API? MQTT? We may want to abstract away the lowest level of the network in favor of providing CPython staples like `requests`.

barlaensdoonn, 2018-05-23:
+1 for this. I'd like to use the Adafruit Feather M0 WiFi w/ATWINC1500 to publish sensor readings via MQTT.

I'd say it's _slightly_  misleading on the M0 WiFi product page to link to a bunch of Circuit Python tutorials when support for the feather's wifi capability doesn't exist.

ladyada, 2018-05-23:
OK! added a note to the product page

> Please note that we do not have CircuitPython support for the WINC1500 modules (it won't fit in the ATSAMD21 chip, not enough room!)

SandeepNandhivardhan, 2018-08-13:
Hi can any give me the binary file and program guide and how to interface ATWINC1500 through AT Commands

ladyada, 2018-08-13:
ATWINC1500 does not use AT commands, it is SPI.

SandeepNandhivardhan, 2018-08-14:
please check on this it has AT Commands
http://ww1.microchip.com/downloads/en/appnotes/atmel-xxxxx-atwinc1500-at-command_applicatio_note.pdf

Can you please show how the binary file can be burned

ladyada, 2018-08-14:
huh, no idea! maybe contact atmel and ask for assistance? we haven't burned AT firmware ever

SandeepNandhivardhan, 2018-08-14:
can you show how it can be programmed using SPI

nickzoic, 2018-08-14:
Hi Sandeep,

I've been working on porting LWIP across to the M4 (SAMD51) port and from there hope to get support for the WizNet 5500 wired ethernet and maybe eventually the ATWINC1500 wifi module too.  But that's a totally different approach.

I had a look at the application note you posted: they look pretty complete and yeah, it looks like once you've loaded some special firmware you can send AT commands to the ATWINC over a UART.  But it seems like a messy way to go about it.

nickzoic, 2018-08-14:
My best suggestion at this point, if you want to use CircuitPython &
WiFi, would be to use a ESP8266  based board.It's a relatively low-end CPU but the WiFi stack is based on LWIP and is
pretty mature.
-----Nick
--
Nick Moore <nick@zoic.org> 0409 656 267

nickzoic, 2018-09-04:
Finally some progress on this!  I've got WizNet5500 support with LWIP for M4 and without LWIP for M0 and M4.  LWIP is a bit of a resource hog so depending on what you're doing the non-LWIP version may work better for you.

Needs to be cleaned up and rebased before PR, but work in progress:
https://github.com/nickzoic/micropython/tree/circuitpython/nickzoic/703-wiznet-5500-samd

Support for WizNet5200 and ATWINC1500 to follow, I hope.

nickzoic, 2018-09-21:
OK so I'm continuing to make progress on moving these across to a "shared-bindings" type of layout for CircuitPython.  The good news is that this will bring networking to other CircuitPython ports as well, really anything which supports SPI should support the W5500 and if its got enough memory then LWIP too.

Progress is at another branch:
https://github.com/nickzoic/micropython/tree/circuitpython/nickzoic/703-wiznet-5500-samd-shared-bindings

turbinenreiter, 2018-11-22:
Hi,

I tried this with the latest alpha. When I run the script from the comment in the Merge Request, I get

* no serial port
* no CIRCUITPY drive

Just wanted to ask what the current status here is and if it's supposed to work, before I file a detailed bug report.

Using an M4 Express with the Ethernet Featherwing.

tannewt, 2018-11-22:
Does USB only fail with the script present? We've had a lot of USB churn recently so it could be unrelated to the network code.

turbinenreiter, 2018-11-22:
Will try to test that tomorrow.

tannewt, 2018-11-24:
We merged some USB fixes today so make sure and update your repo. Thanks!

turbinenreiter, 2018-11-27:
On 027ed82 it's somewhat different.

If I copy-paste above example line-per-line to the REPL, nothing crashes, but I get this error:

```
>>> ss.send("GET / HTTP/1.0\nHost: %s\n\n" % host)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 5] Input/output error
```

... which might not be a problem in the code, but could also be our network. I do get an IP and `True` on `connected` and `dhcp`.

But when I run it from `code.py` it breaks the serial port. When I reset, the storage device shows up, I can connect to the serial, but everything breaks again before I get the chance to do anything.

tannewt, 2018-11-27:
@nickzoic Any ideas?

nickzoic, 2018-11-27:
I'll have a look ...

OK so I've built Adafruit CircuitPython 4.0.0-alpha.3-25-gf6c73b35f-dirty (current master: it's only dirty due to some problem with the nrfx submodule, I'll worry about that later)

The code on the PR works:

```
import board
import busio
import wiznet

spi = busio.SPI(clock=board.SCK, MOSI=board.MOSI, MISO=board.MISO)
eth = wiznet.WIZNET5K(spi, board.D10, board.D11)
eth.connected

host = 'example.com'
import socket
fam, typ, pro, nam, sockaddr = socket.getaddrinfo(host, 80)[0]
ss = socket.socket(fam, typ, pro)
ss.connect(sockaddr)
ss.send("GET / HTTP/1.0\nHost: %s\n\n" % host)
print(ss.recv(1024))
ss.close()
```

I'm using the Metro M4 Express + Ethernet Featherwing too, on top of a custom shield converter which
connects only the necessary pins: GND, 3V, MOSI, MISO, SCLK, and Pin 10 to the CS.

No USB problems for me!  Try powering the M4 from a plugpack in case you're overloading the USB port when the Ethernet module tries to draw power, that can lead to some weird behaviours.

turbinenreiter, 2018-11-28:
* on current master c47c495
* Featherwing on top of the Feather, RST is not connected
* powered from my laptop USB - has not been a problem with the same device when using Arduino

# On the REPL:

```
Adafruit CircuitPython 4.0.0-alpha.3-28-gc47c495ac on 2018-11-28; Adafruit Feather M4 Express with samd51j19
>>> import board
>>> import busio
>>> import wiznet
>>> import socket
>>> spi = busio.SPI(clock=board.SCK, MOSI=board.MOSI, MISO=board.MISO)
>>> eth = wiznet.WIZNET5K(spi, board.D10, board.D11)
>>> eth.connected
True
>>> host = 'example.com'
>>> fam, typ, pro, nam, sockaddr = socket.getaddrinfo(host, 80)[0]
>>> ss = socket.socket(fam, typ, pro)
>>> ss.connect(sockaddr)
>>> ss.send("GET / HTTP/1.0\nHost: %s\n\n" % host)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 5] Input/output error
```
MSD and serial keep working afterwards.

# When I run the script in `code.py`:

```
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Traceback (most recent call last):
  File "code.py", line 10, in <module>
OSError: -2
```
MSD and serial break afterwards.

What puzzles me is that the whole thing behaves differently whether I put it on the REPL or run it as a script. Different error message and different behaviour.

(I tried to build the same commit as @nickzoic but couldn't find gf6c73b35f)

The fact that you can't reproduce is also interesting.

I will try:

* different USB power source
* different network

an report back (probably next week).

Thanks for the support.

nickzoic, 2018-11-29:
OK no worries, I'll try and replicate with that version.

nickzoic, 2018-12-05:
I'll have a look when I find the hardware again!

genevanmeter, 2019-02-12:
> * on current master [c47c495](https://github.com/adafruit/circuitpython/commit/c47c495acaaf1815140571808606c920b23d25b2)
>
>     * Featherwing on top of the Feather, RST is not connected
>
>     * powered [...]

I am having a similar issue were code works in the REPL but if the same code is placed in code.py, the MSD and serial are inaccessible after reset.  Using Feather M4 Express and Ethernet FeatherWing with CircuitPython 4.0.0 Beta 2

nickzoic, 2019-02-12:
> I am having a similar issue were code works in the REPL but if the
> same code is placed in code.py, the MSD and serial are inaccessible
> after reset.  Using Feather M4 Express and Ethernet FeatherWing with
> CircuitPython 4.0.0 Beta 2
Oh, really interesting.  Perhaps the Ethernet board takes a little while
before it's ready to communicate.  I'll check it out.

siddacious, 2019-03-01:
Just chiming in to voice my support for this. I'd be happy to test once the feathers are back in stock.

ladyada, 2019-03-11:
@nickzoic hihi do you have any idea about this or reproduce it?

brentru, 2019-03-11:
I'm encountering a similar issue to the behavior @genevanmeter and @turbinenreiter noticed, on 4.0.0b3. Except, I'm not able to get it to connect.

Running the script in `Code.py` seems to lock up the device (can't view contents in Finder) and the REPL. I had to downgrade to 3.0 to remove the code.py (since it lacks 'wiznet'), then bump back to 4.0 to get the board connecting again.

Calling `.connect`:
```
>>> ss.connect(sockaddr)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: 4
```

Test Setup:
> Adafruit CircuitPython 4.0.0-beta.3 on 2019-03-07; Adafruit Feather M4 Express with samd51j19

nickzoic, 2019-05-02:
@genevanmeter / @turbinenreiter / @brentru / @siddacious please check out #1800 and let me know if that addresses your problem ...

ladyada, 2019-05-02:
@nickzoic please provide UF2's for feather m4 and metro m4 :)

nickzoic, 2019-05-13:
Some outstanding issues were fixed in #1800 and I've added #1522 and #1880 to cover some potential improvements so closing this one at last.

closed 2019-05-13 by nickzoic