adafruit/circuitpython #706: Reproducible assertion failure in usocket.socket() with bad argument list

jepler, 2018-03-25:
Minimized from a testcase found by afl-fuzz:
~~~~
import usocket as socket
s = socket.socket('1')
~~~~

Stack at time of assertion error:
~~~~
micropython: modusocket.c:328: mp_obj_t socket_make_new(const mp_obj_type_t *, size_t, size_t, const mp_obj_t *): Assertion `MP_OBJ_IS_SMALL_INT(args[0])' failed.
==20103== 
==20103== Process terminating with default action of signal 6 (SIGABRT)
==20103==    at 0x5798FCF: raise (raise.c:51)
==20103==    by 0x579A3F9: abort (abort.c:89)
==20103==    by 0x5791E36: __assert_fail_base (assert.c:92)
==20103==    by 0x5791EE1: __assert_fail (assert.c:101)
==20103==    by 0x4E16B9: socket_make_new (modusocket.c:328)
==20103==    by 0x4B41F2: mp_execute_bytecode (vm.c:1014)
~~~~

Upstream micropython gives a Python traceback, not a crash.

tannewt, 2018-03-26:
What version of upstream? We're on 1.9.3 still. (We only update to released MicroPython generally.)

jepler, 2018-03-26:
I only tested with the tip of the mater branch at micropython.

I think the fix is simple, these assertions should just be changed to generate Python exceptions.  I can roll a patch to do that.

This *may* just affect the ports/unix environment used for testing (the filename is ports/unix/modusocket.c so it's not shared with real hardware).  I don't have any CP board with networking so I can't test there.

tannewt, 2018-03-26:
Yeah, first take a look to see if its been fixed up stream.

jepler, 2018-03-31:
@tannewt let's ether close this one or take it off the 3.0 milestone.  It's in ports/unix code, so it can't affect real HW.

tannewt, 2018-04-02:
Happy to close it. Thanks!

closed 2018-04-02 by tannewt