adafruit/circuitpython #707: Reproducible crash concatenating with an O-type array

jepler, 2018-03-25:
Trinket M0, master branch (similar test cases crash interestingly on ports/unix), found with afl-fuzz

~~~~
>>> import array
>>> array.array('O') + array.array('I', [4])
array('O', [
~~~~
Trinket M0 serial interface hangs after printing `[`.

jepler, 2018-03-25:
Python2/3 don't accept array type 'O'.

A related problem can be triggered with slice assignment
~~~~
>>> io = array.array('Q', [2,3,4])
>>> ao[0:2] = io
>>> ao
array('O', ['', 1, Segmentation fault
~~~~

and it's not clear if this result from concatenating different-type arrays is desirable (python2/3 reject it with a typeerror, even when the types have the same size)
~~~~
>>> array.array('B', [2,3,4]) + array.array('I', [5,6,7])
array('B', [2, 3, 4, 5, 0, 0, 0, 6, 0, 0, 0, 7, 0, 0, 0])
~~~~

jepler, 2018-03-26:
Shall we fix slice assignment and concatenation so that different types are rejected, as in Python 2/3? If so I can work on generating a PR.

jepler, 2018-03-26:
Another one to fix: constructing object arrays from bytearrays. 
```
>>> array.array('O', bytearray("baaaaaaa"))array('O', [Segmentation fault
```

dhalbert, 2018-03-26:
A bad typecode should be caught. In CPython 3:
```
>>> array.array('O', [1])
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: bad typecode (must be b, B, u, h, H, i, I, l, L, q, Q, f or d)
```
and I agree slice assignment and concatenation should check that the types match.

Thank you for checking all this! I'm sure upstream (micropython) would be interested as well.

jepler, 2018-03-26:
'O' and 'P' arrays are deliberate MicroPython extensions inherited by CircuitPython—see `tests/basics/array_micropython.py`.

array.array also accepts an 'S' typecode in MicroPython/CircuitPython, but it looks like that one *may* be an accident.  It seems to be in support of the ustruct typecode extension in MicroPython but not in CircuitPython:
```
    This module implements most of character typecodes from CPython, with
    some extensions: …
    S - Pointer to a string (returned as a Python string). Note the
        difference from "Ns", - the latter says "in this place of structure
        is character data of up to N bytes length", while "S" means
        "in this place of a structure is a pointer to zero-terminated
        character data".
```
.. that one we probably should nix from use in array, I think.

jepler, 2018-03-26:
One more:
```
>>> array.array('O') + 'baaaaaaaa'
array('O', [Segmentation fault
```

dhalbert, 2018-03-26:
>'O' and 'P' arrays are deliberate MicroPython extensions inherited by CircuitPython—see tests/basics/array_micropython.py.

Ha! I didn't realize that. I looked in the MPy [readthedocs](http://docs.micropython.org/en/latest/pyboard/library/array.html) and that wasn't listed. I thought `'O'` was just the result of some fuzzing testing.

tannewt, 2018-03-26:
@jepler Overall, we choose to align with CPython instead of MicroPython. So, my vote would be to remove 'O' and 'P' support.

jepler, 2018-03-31:
This was fixed by 355bf8b5538b but apparently I failed to note that it "Closes: ..." in the commit message.

closed 2018-03-31 by jepler