adafruit/circuitpython #714: Prevent unused input characters from spewing into REPL

tannewt, 2018-03-26:
To reproduce:

1. Write a file with an infinite loop but no `input` calls.
2. Type a bunch of characters besides ctrl-c.
3. Type ctrl-c to break out of the loop.
4. The first of a bunch of characters satisfies the any keypress to enter repl and then the repl gets the rest.

Instead, we should clear the input buffer after main finishes and before we wait for another character.

tannewt, 2018-03-29:
@sommersoft This might be a good thing for you since you've poked around the usb stuff.

sommersoft, 2018-03-29:
Yeah. I was already running it in my head.. :smile:

sommersoft, 2018-07-04:
You snuck the 3.x in on me... I'll start working it. 😄

sommersoft, 2018-07-04:
I started reading last night, and got my bearings with `pyexec_nnn_repl`. Started to debug this morning, but I can't seem to reproduce this. Tried a current build (local between alpha.6 and rc.0), as well as alpha.2 and alpha.3 (both before this issue was created).

> Write a file with an infinite loop but no input calls.

I was assuming this is refers to a `code.py/main.py`, and simply put in:
```
while True:
    pass
```
Is that what you meant to reproduce, or do you mean writing _TO_ a file?

tannewt, 2018-07-05:
Yup, thats what I mean. During the infinite loop type some text in then ctrl-c. the text you typed in before will spill into the repl.

I think the fix is to clear the USB buffer from main.c after code.py finishes.

sommersoft, 2018-07-05:
After multiple attempts, I still cannot replicate this behavior. Closing for now; will reopen if it occurs in the future.

- Boards used to attempt replication: Metro M4 Express (Beta), Trinket M0.
- Firmware tested: 3.alpha-2, 3.alpha-3, and 3.rc-1.
- REPL Terminal Programs (Windows): PuTTY & Mu REPL

closed 2018-07-05 by sommersoft