adafruit/circuitpython #716: PulseIn not implemented for ESP8266

jerryneedell, 2018-03-26:
When trying to use a DHT22 on an ESP8266 - this error is returned:
```
>>> import dht22_test
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "dht22_test.py", line 7, in <module>
  File "adafruit_dht.py", line 190, in temperature
  File "adafruit_dht.py", line 143, in measure
  File "adafruit_dht.py", line 106, in _get_pulses
NotImplementedError: 
```

https://github.com/adafruit/Adafruit_CircuitPython_DHT/blob/master/adafruit_dht.py#L106

I did not see an issue for this so I am just creating this to track it.

jerryneedell, 2018-03-26:
It is also not implemented for the nrf52. Changed title.

nickzoic, 2018-06-06:
I've started working on this over at https://github.com/nickzoic/micropython/tree/circuitpython-nickzoic-716-pulseio-esp8266 ... still super sketchy at the moment and missing a lot of the API features.

tannewt, 2018-06-07:
@nickzoic Looks like a good start! You may be interested in my pending changes here too: https://github.com/adafruit/circuitpython/compare/master...tannewt:pulseio_too_fast

It's adding some handling for gracefully failing when the input signal causes the interrupt to swamp the main code.

nickzoic, 2018-06-07:
Interesting.  I'm not sure if there's an equivalent for the '66 port, but there's possibly less need for it since communications are UART based ... 
I was wondering about the pulse buffer overflow condition: perhaps it'd make more sense to pause if the ring buffer overflows and resume when you pop the ring buffer. At the moment it over-writes the earliest pulses, but quite often you're interested in the start of a packet ... and it'd also solve the swamping problem.

nickzoic, 2018-06-08:
OK, PulseIn in that branch is looking pretty good now, and can reliably receive IR remote control signals and can be used by the adafruit_dht library to interface to a DHT11.

PulseOut is on ongoing effort mostly because of the terrible state of ESP8266 PWM and microsecond timers.

dhalbert, 2018-07-17:
split off nrf52 issue into #1043.

sommersoft, 2018-07-17:
Closing since ESP8266 is fixed by #926, and nRF is split out.

Thanks @nickzoic!

closed 2018-07-17 by sommersoft