adafruit/circuitpython #717: LORA Radio driver fails under CP3.0

jerryneedell, 2018-03-26:
Running the basic test fails under CP3..0 alpha (current master)
It works under CP 2.2.4

Same boards, only change is reloading CP and libraries.

```
Adafruit CircuitPython 3.0.0-alpha.3-31-g8b6aeb9-dirty on 2018-03-26; Adafruit Feather M0 Express with samd21g18
>>> import lora_test
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "lora_test.py", line 26, in <module>
  File "adafruit_rfm9x.py", line 363, in __init__
  File "adafruit_rfm9x.py", line 361, in __init__
RuntimeError: Failed to configure radio for LoRa mode, check wiring!
>>> 

```
lora_test.py is just the repo simpletest with the correct pins set.
```
# Simple demo of sending and recieving data with the RFM95 LoRa radio.
# Author: Tony DiCola
import board
import busio
import digitalio

import adafruit_rfm9x

# Define radio parameters.
RADIO_FREQ_MHZ   = 915.0  # Frequency of the radio in Mhz. Must match your
                          # module! Can be a value like 915.0, 433.0, etc.

# Define pins connected to the chip, use these if wiring up the breakout according to the guide:
CS    = digitalio.DigitalInOut(board.D10)
RESET = digitalio.DigitalInOut(board.D11)
# Or uncomment and instead use these if using a Feather M0 RFM9x board and the appropriate
# CircuitPython build:
#CS    = digitalio.DigitalInOut(board.RFM9X_CS)
#RESET = digitalio.DigitalInOut(board.RFM9X_RST)

# Initialize SPI bus.
spi = busio.SPI(board.SCK, MOSI=board.MOSI, MISO=board.MISO)

# Initialze RFM radio
rfm9x = adafruit_rfm9x.RFM9x(spi, CS, RESET, RADIO_FREQ_MHZ)

# Note that the radio is configured in LoRa mode so you can't control sync
# word, encryption, frequency deviation, or other settings!

# You can however adjust the transmit power (in dB).  The default is 13 dB but
# high power radios like the RFM95 can go up to 23 dB:
rfm9x.tx_power = 23

# Send a packet.  Note you can only send a packet up to 252 bytes in length.
# This is a limitation of the radio packet size, so if you need to send larger
# amounts of data you will need to break it into smaller send calls.  Each send
# call will wait for the previous one to finish before continuing.
rfm9x.send('Hello world!\r\n')
print('Sent hello world message!')

# Wait to receive packets.  Note that this library can't receive data at a fast
# rate, in fact it can only receive and process one 252 byte packet at a time.
# This means you should only use this for low bandwidth scenarios, like sending
# and receiving a single message at a time.
print('Waiting for packets...')
while True:
    packet = rfm9x.receive()
    # Optionally change the receive timeout from its default of 0.5 seconds:
    #packet = rfm9x.receive(timeout_s=5.0)
    # If no packet was received during the timeout then None is returned.
    if packet is None:
        print('Received nothing! Listening again...')
    else:
        # Received a packet!
        # Print out the raw bytes of the packet:
        print('Received (raw bytes): {0}'.format(packet))
        # And decode to ASCII text and print it too.  Note that you always
        # receive raw bytes and need to convert to a text format like ASCII
        # if you intend to do string processing on your data.  Make sure the
        # sending side is sending ASCII data before you try to decode!
        packet_text = str(packet, 'ascii')
        print('Received (ASCII): {0}'.format(packet_text))
        # Also read the RSSI (signal strength) of the last received message and
        # print it.
        rssi = rfm9x.rssi
        print('Received signal strength: {0} dB'.format(rssi))

```

jerryneedell, 2018-05-02:
I have confirmed that this issue is still present with CP 3.0 Master. The first check of the configuration fails
https://github.com/adafruit/Adafruit_CircuitPython_RFM9x/blob/master/adafruit_rfm9x.py#L360
This appears to be a problem with basic write/read.
Note - the driver works under CP 2.2.4  and the RFM69 driver works under CP3.0 master.
I have also experienced several instances where the FIle System is corrupted while troubleshooting this.
I will hook up a Logic analyzer next and look at the write/read operations.

jerryneedell, 2018-05-02:
making progress:  setting the RESET to Pull.UP  at initialization fixes the driver for CP 3.0

```
diff --git a/adafruit_rfm9x.py b/adafruit_rfm9x.py
index d5afcef..72d5d93 100644
--- a/adafruit_rfm9x.py
+++ b/adafruit_rfm9x.py
@@ -30,7 +30,7 @@ http: www.airspayce.com/mikem/arduino/RadioHead/
 * Author(s): Tony DiCola
 """
 import time
-
+import digitalio
 from micropython import const
 
 import adafruit_bus_device.spi_device as spi_device
@@ -344,7 +344,7 @@ class RFM9x:
         # trigger a reset.  Note that reset MUST be done like this and set as
         # a high impedence input or else the chip cannot change modes (trust me!).
         self._reset = reset
-        self._reset.switch_to_input()
+        self._reset.switch_to_input(pull=digitalio.Pull.UP)
         self.reset()
         # No device type check!  Catch an error from the very first request and
         # throw a nicer message to indicate possible wiring problems.
@@ -421,7 +421,7 @@ class RFM9x:
         # See section 7.2.2 of the datasheet for reset description.
         self._reset.switch_to_output(value=False)
         time.sleep(0.0001)  # 100 us
-        self._reset.switch_to_input()
+        self._reset.switch_to_input(pull=digitalio.Pull.UP)
         time.sleep(0.005)   # 5 ms
 
     def idle(self):
```
I'm not sure what is different between 2.x in an 3,0, but the RESET was not working when left to the default..

I want to play around with the RESET sequence before submitting a PR but it is nice to have it working!

jerryneedell, 2018-05-03:
submitted PR https://github.com/adafruit/Adafruit_CircuitPython_RFM9x/pull/2  that , I hope, takes care of this.  In any case, it appears to be a driver issue and not a CP issue.
OK to close?

jerryneedell, 2018-05-03:
closed via https://github.com/adafruit/Adafruit_CircuitPython_RFM9x/pull/2

closed 2018-05-03 by jerryneedell

MakabongweNkabinde, 2019-10-05:
Hi there

I'm having a similar issue. I have connected the RFM96W according to the tutorial here: https://learn.adafruit.com/lora-and-lorawan-radio-for-raspberry-pi/raspberry-pi-wiring. I have two of these modules, one of them is detected but one of them keeps spitting out this error. Is it possible that the module is broken? How can I test if the module is actually broken? Could it be another issue? Your help would be appreciated. Thanks.

jerryneedell, 2019-10-05:
are you using the rfm96w breakout? https://www.adafruit.com/product/3073
Since one works, it is likely not a driver or CP issue.
First, have you checked the solder connections? 
Also what pins are you using for Chip Select --when you swap modules are you using the same connections for both or is it on two different Raspberry Pi's

jerryneedell, 2019-10-05:
Alos - since this is probably not a CP issue and may be a hardware problem I suggest that you post this on the Forum at https://forums.adafruit.com/viewforum.php?f=57  -- we can continue there and include the Adafruit Support folks.

MakabongweNkabinde, 2019-10-05:
Hi jerryneedell

Thanks for the response. No I'm not using the rfm96w breakout, I'm using the chip as it is.  I have checked the connections. I am using CE1 for chip select; it's two different raspberry pi's and I use the sane connections I used on the other pi.

jerryneedell, 2019-10-05:
Can you swap the module with the working one on the other Pi?

jerryneedell, 2019-10-05:
You may want to post this to the discord channel for radio discussion, https://adafru.it/discord  then go to the #help-with-radio channel

jonathanmccormick, 2019-12-10:
Hey @Makabongwe-Nkabinde , did you ever get this figured out? I'm having the same problem. Triple checked my wiring...

jerryneedell, 2019-12-10:
@jonathanmccormick please provide more details about your issue. What boards are you using. What code? Are you using the latest libraries. Are you using a raspberry pi or if not, what MCU and what version of CIrcuitPython.  I have been running these libraries successfully.

Makabongwe-Nkabinde, 2019-12-10:
@jonathanmccormick the issue was with the chip itself, I bought a new one and that solved the issue. The other issue could could be the interrupt request pin, make sure that is configured correctly.