adafruit/circuitpython #718: DHT22 driver not working under CP3.0-alpha

jerryneedell, 2018-03-26:
dht_simpletest yields:
```
Adafruit CircuitPython 3.0.0-alpha.3-31-g8b6aeb9-dirty on 2018-03-26; Adafruit Feather M0 Express with samd21g18
>>> 
>>> 
>>> import dht_simpletest
('A full buffer was not returned.  Try again.',)
('A full buffer was not returned.  Try again.',)
('A full buffer was not returned.  Try again.',)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "dht_simpletest.py", line 18, in <module>
KeyboardInterrupt: 
>>> 
```

under CP3.0 alpha  
It works ok on 2.2.4.

sommersoft, 2018-03-29:
Looking into this real quick. Not to state the obvious, but the `pulsein` is failing to return any data. I turned off the `raise RuntimeError` in the driver, and enabled the print statement that was commented out. Here is the return message:
```
did not get a full return.  number returned was: 0
```

Will dig a little deeper.

sommersoft, 2018-03-31:
Update on things I've poked at so far:

- Changing the `trigger_duration` that is called from the driver. Its set to `1000` for DHT22, and `18200` for DHT11 (don't have an 11; how could this not cause a lockup with `delay_us`?). Setting the DHT22 above `1000` causes a lockup. Anything below `1000` yields the current result of no readings.

- Re-ordered some of the `EIC/GCLK` calls in `pulsein_resume` and `pulsein_construct`. No changes to results.

- Verified _most_ of the struct variables used in `EIC` & `PM`. These all look good so far.

I'll keep plugging away, but these be uncharted waters for me...

sommersoft, 2018-04-01:
Slight progress. I think. 

I've stepped through several function breakpoints, several times. While chasing variables and doing my best to verify them, I noticed that `EIC_Handler -> pulsein_interrupt_handler` is only called on the `first_edge` response. It never triggers beyond that.

I tried changing the `idle_state` to `false` in the driver, since the first data response _should_ be a rising edge, not a falling, according to the datasheet if I'm reading it right. Didn't make a difference, since the interrupt handler is only called once.

@tannewt, any suggestions on what may be causing this? Or am I totally not understanding how the interrupt handler should be acting?

tannewt, 2018-04-02:
I think you are on the right track. The interrupt should fire on the specific first edge and then fire on both. This can be very tricky to debug since the DHT will return the data even if you are stopped at a breakpoint.

I'd suggest having a Saleae to validate that the sensor is responding OK. (Ping me if you don't have one.) That can also be useful to validate the measurements against a "neutral third party" so to speak.

Other than that, I don't have any specific ideas.

jerryneedell, 2018-04-04:
I'm finally back home and will be happy to help with this. I feel somewhat responsible ;-)

sommersoft, 2018-04-09:
Not-much-of-an-update, update.

After much reading, and comparing to 2.x/ASF3, I'm not exactly any closer. I may have gotten _somewhat_ closer, since I have determined that the previously mentioned calls to `EIC_Handler` are actually being called during `common_hal_pulseio_pulsein_construct` and not during the `resume` call. After seeing that, I started playing around with two things:

1. `pulsein_set_config` - I turned off the disabling/enabling of the EIC to set the config. I know that the SAMD51 datasheet says that the `EIC_CONFIG` registers are "Enable-Protected", but the SAMD21 doesn't mention it. Furthermore, 2.x/ASF3 doesn't disable the EIC when updating the config; it calls [`extint_chan_disable_callback`](https://github.com/adafruit/circuitpython/blob/2.x/atmel-samd/asf/sam0/drivers/extint/extint_callback.c#L194), which only sets the `INTCLR` bit. The only difference in leaving the EIC turned on during `pulsein_set_config` was it expanded the returns of `EIC_Handler` (by 1; see logs below).

2. Explicitly setting the pin direction and pull during `common_hal_pulseio_pulsein_construct` and after the trigger duration section in `common_hal_pulseio_pulsein_resume`. I couldn't determine that setting the pin's function with `gpio_set_pin_funtion` was doing this. With the `construct` triggering `EIC_Handler`, I was suspicious that the pin may actually being sending a signal OUT and triggering the DHT. I noticed no difference with adding explicit direction/pull.

That's about as far as I've gotten to this point. Here are some debug logs illustrating what I'm seeing based on the callstacks. I have gotten these results consistently, so I don't think it's a matter of Atmel Studio not being able to keep up.

```
## eic_set_enable ON in pulsein_set_config ##
---------------------------------------------------
pulsein_interrupt_handler(EIC_Handler):: last_ms: 0; last_us: 0; 
        firmware.elf! external_interrupt_handler
	firmware.elf! EIC_Handler
	firmware.elf! <signal handler called>
	firmware.elf! common_hal_pulseio_pulsein_construct
	firmware.elf! heap
```
```
## eic_set_enable OFF in pulsein_set_config ##
----------------------------------------------------
pulsein_interrupt_handler(EIC_Handler):: last_ms: 0; last_us: 0; 	
        firmware.elf! external_interrupt_handler
	firmware.elf! EIC_Handler
	firmware.elf! <signal handler called>
	firmware.elf! common_hal_pulseio_pulsein_construct
	firmware.elf! pulseio_pulsein_make_new
	firmware.elf! type_call
	firmware.elf! mp_call_function_n_kw
	firmware.elf! mp_call_method_n_kw
	firmware.elf! mp_execute_bytecode
        ....# TRUNCATED FOR BREVITY #....
	
pulsein_interrupt_handler(EIC_Handler):: last_ms: 266659; last_us: 400;
        firmware.elf! external_interrupt_handler
	firmware.elf! EIC_Handler
	firmware.elf! <signal handler called>
	firmware.elf! eic_set_enable
	firmware.elf! common_hal_pulseio_pulsein_construct
	firmware.elf! pulseio_pulsein_make_new
	firmware.elf! type_call
	firmware.elf! mp_call_function_n_kw
	firmware.elf! mp_call_method_n_kw
	firmware.elf! mp_execute_bytecode
        ....# TRUNCATED FOR BREVITY #....
```

As you can see in the second log, an additional interrupt was handled and shows that `pulsein_interrupt_handler` is at least functioning by updating the ms/us vars. Also, I didn't put it in my logs, but `common_hal_pulseio_pulsein_resume` would show up in the callstack AFTER the above interrupts.

tannewt, 2018-04-09:
I have had the problem before where I enable an interrupt without clearing its previous state. We should also make sure we turn the interrupt off while we configure things.

Sorry, I don't have more to offer. It sounds like you are on the right track so I haven't taken a look at it. Thanks!

sommersoft, 2018-04-09:
> I have had the problem before where I enable an interrupt without clearing its previous state. We should also make sure we turn the interrupt off while we configure things.

Untangling the NVIC and EIC interaction has been most of this week's work. If I'm understanding it correctly, things are occuring as they should.

sommersoft, 2018-04-14:
Saleae Update.

After learning to use it (short learning curve...Saleae did excellent on that front), here is the 2.x to 3.x comparison. 

**2.2.4**
[image]

**3.0.0-alpha.4**
[image]

Based on the duration annotations, I think we're shorting the initial start signal. Depending on which datasheet you read, start signal duration must be either 500us or 1ms. However, using either value, it's clear we're short on `3.x`. But, as I mentioned above, setting the `trigger_duration > 1000` in the driver itself causes a lockup.

Nice to have physical data though to narrow the focus.

jerryneedell, 2018-04-14:
setting it to 500 - I got a few more readings - but the SALEA only show a 4.6 microsecond pulse not 500 microsecond.

sommersoft, 2018-04-14:
@jerryn, did the Saleae show more than the first low pulse? Wondering if your catching the 2 interrupts I was getting from `construct`...

jerryneedell, 2018-04-14:
this was with 500  -- I see the pushes, but am getting bad checksums most of the time...
[image]

jerryneedell, 2018-04-14:
at a triger_duration of 100 - I do see a 100 microsecond pulse and a good first reading. then a bad checksum on the next.

jerryneedell, 2018-04-15:
@sommersoft I think the reason it locks up for tick_duration > 1000 is becasue it is usint tick_delay with interrupts disabled - this will wait forever since tick_ms won't increment with interrupts disabled.

https://github.com/adafruit/circuitpython/blob/master/ports/atmel-samd/tick.c#L64

jerryneedell, 2018-04-15:
This function does not appear to be generating consistent delays.
https://github.com/adafruit/circuitpython/blob/master/ports/atmel-samd/tick.c#L59

jerryneedell, 2018-04-15:
Still trying to understand tick_delay, but this is all new to 3.0 and it look like the problem is caused here.  If the requested delay is less than the time until the next tick, I think the current logic generates a short pulse for the Pulseio.resume trigger_duration.

sommersoft, 2018-04-15:
@jerryneedell that is exactly where I was heading too. I was studying it yesterday. I'm going to start _trying_ to read some values during debug...

jerryneedell, 2018-04-15:
@sommersoft I tried simply replacing the call to tick_delay with delay_us  (as in 2.2.4)  and setting the dely back to 1000. It now runs but I get quite a few checksum errors about and bad readings. However, I do get many successful readings.  Not a real fix, but I think it points to tick_delay as the main culprit.

jerryneedell, 2018-04-15:
```
Adafruit CircuitPython 3.0.0-alpha.4-dirty on 2018-04-15; Adafruit Feather M0 Express with samd21g18
>>> 
>>> 
>>> import dht22_test
20.6 55.0
20.6 54.9
20.6 54.9
 RunTime Error  Checksum did not validate. Try again.
225.4 54.8
 RunTime Error  Checksum did not validate. Try again.
225.4 54.8
 RunTime Error  Checksum did not validate. Try again.
225.4 54.8
20.6 54.7
20.6 54.7
20.6 54.7
 RunTime Error  Checksum did not validate. Try again.
6553.5 6553.5
20.6 54.6
20.6 54.6
 RunTime Error  Checksum did not validate. Try again.
225.4 54.6
20.6 54.7
 RunTime Error  Checksum did not validate. Try again.
225.4 54.8
20.6 54.9
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "dht22_test.py", line 11, in <module>
  File "dht22_test.py", line 9, in <module>
KeyboardInterrupt: 
>>> 
```
using
```
from board import D13
import adafruit_dht as dht
import time

sensor=dht.DHT22(D13)
while True:
    try:
        print(sensor.temperature,sensor.humidity)
        time.sleep(1)
    except RuntimeError as e:
        print(" RunTime Error ", e)

```

jerryneedell, 2018-04-15:
There may be 2 issues going on. If I go back to using tick_delay with the code ablove, it eventaully recovers from the incomplete buffer errors then gives some good readings and some bad readings.

jerryneedell, 2018-04-15:
@sommersoft#0222  I think there are 2 problems with tick_delay --  as written ```
us_between_ticks - us
``` 
will be negative since us is 1000 in our case and us_between_tics is always < 1000  but I think since us_between_ticks and us ar defines as unsigned ints,  this diff is always assignee 0xffffffff  because the diff is negative - so tick_delay returns immediately. Setting us to < 1000  will sometimes work as seen depending if us_between_ticks is > us.

If we "fix this" then it will hang because SysTick->VAL is not reset unless interrupts are enabled.

jerryneedell, 2018-04-15:
example of mis handling of diff between unsigned ints.

```
jerryneedell@Ubuntu-Macmini:~/projects/metro_m4_express$ ./test
64 a 5a ffffffa6
jerryneedell@Ubuntu-Macmini:~/projects/metro_m4_express$ cat test.c
#include <stdint.h>
#include <stdio.h>
void main()
{
    uint32_t x,y;

    x=100;
    y=10;
    printf("%x %x %x %x\n",x,y,x-y,y-x);
}
```

sommersoft, 2018-04-15:
@jerryneedell it seems that this is similar to the 2.x `wait_until` function, in that it too would freeze with interrupts off. I finally was able to step through `tick_delay` with debug (got atmel studio to force it with debug set). Here is what I saw:

```
tick_ms: 199340
ticks_per_us: 48      # (obviously, but nice to verify)
us_between_ticks: 178
start_ms: <optimized out> unfortunately; even though it isn't used in the final calc?
```

which make the final eval = `while (SysTick->VAL > ((178 - 1000) * 48)) {}`. So, like you said, `SysTick->VAL` will have to somehow equal -47,822.

jerryneedell, 2018-04-15:
right - I see the same thing, but as an unsigned int, -47,822 is alway greater than SysTick->Val so it will return immediately thus the short pulse.  Id er set trigger_duration to say 500 then sometime we get lucky and we get a normal start pulse. Sometimes not depending on where the call comes with respire tot the 1ms tick cycle.

jerryneedell, 2018-04-15:
OK - finally made some progress - I changed tick_delay to:
```
void tick_delay(uint32_t us) {
    uint32_t ticks_per_us = common_hal_mcu_processor_get_frequency() / 1000 / 1000;
    uint32_t us_between_ticks = SysTick->VAL / ticks_per_us;
    uint64_t start_ms = ticks_ms;
    while (us > 1000) {
        while (ticks_ms == start_ms) {}
        us -= us_between_ticks;
        start_ms = ticks_ms;
        us_between_ticks = 1000;
    }
    if(us < us_between_ticks){
        while (SysTick->VAL > ((us_between_ticks - us) * ticks_per_us)) {}
    }
    else {
        while (SysTick->VAL < (999 * ticks_per_us)) {}
        us -= us_between_ticks;
        while (SysTick->VAL > ((1000 - us) * ticks_per_us)) {}
    }
}
```

The 999 is a bit of a kludge, but it works for now- This eliminates the "not full buffer returned" errors and sends a consistent pulse of 1000microseconds  (note you still cannot set trigger_duration > 1000 or it will  still hang) I think I can deal with that as well using a similar kludge but I wanted to get the DHT22 working first.

I still get a mixture of good readings and checksum failures on the feather m0. I get all checksum errors on the metro_m4.  I suspect this a due to errors decoding the pulse durations used to reads 0s/1s from the dht22. This may also be related to the way timing is done under 3.0.  Here is an exampel running the same test code now on the feather_m0 with the updated tick_delay.

```
Adafruit CircuitPython 3.0.0-alpha.4-dirty on 2018-04-15; Adafruit Feather M0 Express with samd21g18
>>> 
>>> import dht22_test
21.6 54.5
 RunTime Error  Checksum did not validate. Try again.
6553.5 6553.5
21.5 54.4
 RunTime Error  Checksum did not validate. Try again.
6553.5 6553.5
 RunTime Error  Checksum did not validate. Try again.
6553.5 6553.5
 RunTime Error  Checksum did not validate. Try again.
6553.5 6553.5
21.6 54.4
21.6 54.5
21.6 54.5
21.6 54.6
21.6 54.6
21.6 54.6
 RunTime Error  Checksum did not validate. Try again.
6553.5 6553.5
21.6 54.7
21.6 54.6
 RunTime Error  Checksum did not validate. Try again.
6553.5 6553.5
 RunTime Error  Checksum did not validate. Try again.
21.6 56.2
21.6 54.6
21.6 54.6
21.6 54.6
```
Checking with  the SALEA, the start pulse seems to be consistently at 1000 microsec. even when tere are checksum errors.

jerryneedell, 2018-04-15:
this may be hard to read, but it shows the pulses and decoded values for some dht22 reads.  You can see that some of the reads have a spurious pulse of lengh  80 at the start and this trows off the whole train.

Others have some clearly bad pulse lengths (>1000)
```
Adafruit CircuitPython 3.0.0-alpha.4-dirty on 2018-04-15; Adafruit Feather M0 Express with samd21g18
>>> 
>>> import dht22_test
array('H', [53, 26, 54, 26, 54, 26, 54, 26, 53, 27, 53, 26, 54, 74, 54, 25, 66, 26, 54, 26, 54, 74, 53, 74, 54, 73, 54, 26, 54, 73, 54, 73, 67, 26, 53, 27, 53, 26, 54, 26, 54, 26, 54, 26, 53, 26, 54, 25, 67, 74, 53, 74, 54, 26, 54, 73, 54, 26, 54, 74, 53, 74, 54, 72, 64, 27, 53, 26, 54, 26, 54, 73, 54, 26, 54, 74, 53, 26, 54, 25, 47])
array('B', [2, 59, 0, 215, 20])
21.5 57.1
array('H', [80, 54, 26, 54, 26, 53, 27, 53, 26, 54, 26, 54, 26, 53, 74, 54, 25, 67, 26, 53, 26, 54, 74, 53, 26, 54, 74, 54, 26, 53, 26, 54, 25, 67, 26, 54, 26, 53, 26, 54, 26, 54, 26, 54, 26, 53, 27, 53, 25, 67, 74, 53, 74, 54, 26, 54, 73, 54, 74, 54, 26, 53, 27, 53, 25, 64, 27, 53, 26, 54, 26, 54, 26, 54, 26, 53, 26, 54, 74, 54, 25])
array('B', [255, 255, 255, 255, 255])
 RunTime Error  Checksum did not validate. Try again.
6553.5 6553.5
array('H', [54, 26, 54, 26, 53, 27, 53, 26, 54, 26, 54, 26, 54, 73, 54, 25, 67, 26, 54, 26, 53, 74, 54, 26, 54, 26, 53, 74, 54, 74, 53, 73, 67, 26, 53, 27, 53, 26, 54, 26, 54, 26, 54, 26, 53, 26, 54, 25, 67, 74, 53, 74, 54, 26, 54, 73, 54, 74, 53, 27, 53, 26, 54, 25, 64, 26, 54, 26, 54, 26, 54, 26, 53, 26, 54, 26, 54, 26, 54, 72, 48])
array('B', [2, 39, 0, 216, 1])
21.6 55.1
array('H', [53, 26, 54, 26, 54, 26, 54, 26, 54, 26, 53, 26, 54, 74, 54, 25, 66, 26, 54, 26, 54, 74, 53, 26, 54, 26, 54, 74, 53, 74, 54, 73, 66, 26, 54, 26, 54, 26, 54, 26, 53, 26, 54, 26, 54, 26, 54, 25, 66, 74, 54, 73, 54, 26, 54, 74, 53, 74, 54, 26, 54, 26, 53, 26, 64, 26, 53, 27, 53, 26, 54, 26, 54, 26, 54, 26, 54, 26, 53, 73, 47])
array('B', [2, 39, 0, 216, 1])
21.6 55.1
array('H', [53, 26, 54, 26, 54, 26, 54, 26, 53, 26, 54, 26, 54, 74, 53, 25, 67, 26, 54, 26, 54, 73, 54, 26, 54, 26, 54, 73, 54, 74, 53, 73, 67, 26, 54, 26, 53, 27, 53, 26, 54, 26, 54, 26, 53, 27, 53, 25, 67, 74, 54, 73, 54, 26, 54, 73, 54, 74, 54, 26, 53, 26, 54, 25, 64, 26, 54, 26, 54, 26, 54, 26, 53, 26, 54, 26, 54, 26, 54, 72, 48])
array('B', [2, 39, 0, 216, 1])
21.6 55.1
array('H', [54, 26, 54, 26, 54, 26, 53, 26, 54, 26, 54, 26, 54, 73, 54, 25, 67, 26, 54, 26, 53, 74, 54, 26, 54, 26, 53, 74, 54, 74, 53, 73, 67, 26, 53, 27, 53, 26, 54, 26, 54, 26, 54, 26, 53, 26, 54, 25, 67, 74, 53, 74, 54, 26, 54, 73, 54, 74, 53, 26, 54, 26, 54, 25, 64, 26, 54, 26, 54, 26, 54, 26, 53, 27, 53, 26, 54, 26, 54, 72, 48])
array('B', [2, 39, 0, 216, 1])
21.6 55.1
array('H', [54, 26, 54, 26, 53, 27, 53, 26, 1054, 26, 54, 26, 53, 74, 54, 25, 67, 26, 53, 27, 53, 74, 54, 26, 53, 27, 53, 74, 54, 73, 54, 73, 67, 26, 53, 26, 54, 26, 54, 26, 54, 26, 53, 26, 54, 26, 54, 25, 67, 73, 54, 74, 54, 26, 53, 74, 54, 74, 53, 26, 54, 26, 54, 25, 64, 26, 54, 26, 54, 26, 54, 1026, 53, 27, 53, 26, 54, 26, 54, 73, 47])
array('B', [2, 39, 0, 216, 17])
 RunTime Error  Checksum did not validate. Try again.
21.6 55.1
array('H', [54, 26, 54, 26, 53, 26, 54, 26, 54, 26, 54, 26, 53, 74, 54, 25, 67, 26, 53, 26, 54, 74, 53, 27, 53, 26, 54, 74, 53, 74, 54, 73, 66, 26, 54, 26, 54, 26, 53, 27, 53, 26, 54, 26, 54, 26, 53, 25, 67, 74, 54, 73, 54, 26, 54, 74, 53, 74, 54, 26, 54, 26, 53, 25, 65, 26, 53, 27, 53, 26, 1054, 26, 54, 26, 54, 26, 53, 26, 54, 73, 47])
array('B', [2, 39, 0, 216, 1])
21.6 55.1
```

jerryneedell, 2018-04-15:
ah - fixed the metro_m4 and most of the errors on the feather_m0:
in the dht driver:
```
    def _get_pulses(self):
        """ _get_pulses implements the communication protcol for
        DHT11 and DHT22 type devices.  It sends a start signal
        of a specific length and listens and measures the
        return signal lengths.

        return pulses (array.array uint16) contains alternating high and low
        transition times starting with a low transition time.  Normally
        pulses will have 81 elements for the DHT11/22 type devices.
        """
        pulses = array.array('H')
        tmono = time.monotonic()

        # create the PulseIn object using context manager
        with pulseio.PulseIn(self._pin, 81, True) as pulse_in:

            # The DHT type device use a specialize 1-wire protocol
            # The microprocessor first sends a LOW signal for a
            # specific length of time.  Then the device sends back a
            # series HIGH and LOW signals.  The length the HIGH signals
            # represents the device values.
            pulse_in.pause()
            pulse_in.clear()
            pulse_in.resume(self._trig_wait)

            # loop until we get the return pulse we need or
            # time out after 1/2 seconds
            while True:
                if len(pulse_in) >= 80:
                    break
                if time.monotonic()-tmono > 0.5: # time out after 1/2 seconds
                    break

            pulse_in.pause()
            while pulse_in:
                pulses.append(pulse_in.popleft())
            pulse_in.resume()

        return pulses

```

keeps the last 81 pulses received  but it is cutting off a bit too soon If I change:
```
         while True:
                if len(pulse_in) >= 80:
                    break
                if time.monotonic()-tmono > 0.5: # time out after 1/2 seconds
                    break

```
to 
```
           while True:
                if len(pulse_in) >= 82:
                    break
                if time.monotonic()-tmono > 0.5: # time out after 1/2 seconds
                    break

```
it works much better. The reason is that the dht 22 puts out a few starting pulses of 20,80,800 microseconds before the real stream of data begins. We only want the final 81 pulses. By waiting longer we skip the startup bits.  This may need some tweaking  still but it makes a big difference. We want to get the right number of bits without having to wait for the timeout.

jerryneedell, 2018-04-15:
example from metro_m4_express_revb  - with modified tick_delay and modified dht driver.
Still some glitches and I had some cases where I think the "edge condition" noted above cause some problems.  Overall - it is "working" under 3.0 - not perfect, but much better.  
If you think this is headed in the right direction I'll create a PR tomorrow,

jerryneedell, 2018-04-15:
I think I have a much cleaner way to detect the reset of SysTick->VAL. I’ll try it tomorrow. I’ll just look for it to increase in value. This will signify a reset.  It won’t be a big change so let me know if you have other suggestions.

tannewt, 2018-04-16:
@sommersoft and @jerryneedell you are amazing! Thank you for sorting this out!

I don't think we need another issue. This has so much good background it'd be worse to split the discussion up. Thanks again!

jerryneedell, 2018-04-16:
OK - I think this fixes tick_delay and allow it to work even for us > 1000.
```
void tick_delay(uint32_t us) {
    uint32_t ticks_per_us = common_hal_mcu_processor_get_frequency() / 1000 / 1000;
    uint32_t us_between_ticks = SysTick->VAL / ticks_per_us;
    uint32_t start_tick;
    while (us > 1000) {
        start_tick=SysTick->VAL;  // wait for SysTick->VAL to  RESET
        while (SysTick->VAL < start_tick) {}
        us -= us_between_ticks;
        us_between_ticks = 1000;
    }
    if(us&&(us < us_between_ticks)){
        while (SysTick->VAL > ((us_between_ticks - us) * ticks_per_us)) {}
    }
    else {
        start_tick=SysTick->VAL;  // wait for SysTick->VAL to  RESET
        while (SysTick->VAL < start_tick) {}
        us -= us_between_ticks;
        if(us){
            while (SysTick->VAL > ((1000 - us) * ticks_per_us)) {}
        }
    }
}

```
I'll put in a PR later today.  

Also finishing up some tweaks to the dht driver.  It is now working on both the Metro_M4 and feather_M0.  Still get some checksum failures, but they may just be a fact of lif for the DHT22 - I have always seen them in the past as well. Also need to try the DHT11

jerryneedell, 2018-04-21:
I think this is fixed by #765 and by DHT driver update (pr_8). 
There are occasional checksum failures. And I want to dig into pulsio more. Should we track it here or close this and open a new issue specifically for the DHT checksum errors.

tannewt, 2018-04-23:
@jerryneedell good work! New issue for checksum errors would be good.

closed 2018-04-23 by tannewt