adafruit/circuitpython #725: Not able to build

srfoster, 2018-03-29:
I think I'm following the README correctly, but maybe not.  In the atmel-samd directory, I ran

'''
make BOARD=circuitplayground_express
'''

Get a few hundred errors that all look basically like this:

'''
compilation terminated.
In file included from ../../py/mpconfig.h:45:0,
                 from ../../py/lexer.h:31,
                 from ../../extmod/vfs.h:29,
                 from ../../shared-module/os/__init__.c:31:
./mpconfigport.h:137:10: fatal error: include/sam.h: No such file or directory
 #include "include/sam.h"
          ^~~~~~~~~~~~~~~
'''
Am I missing a dependency?  What is sam.h?

sommersoft, 2018-03-29:
did you run `git submodule update --init --recursive` after cloning the repository? also need to run `make mpy-cross`.

sommersoft, 2018-03-29:
And yes, it is missing from the README...

srfoster, 2018-03-29:
Wow.  That's the fastest anyone has responded to a github issue ever.  Thanks!

I did build mpy-cross. I hadn't run `git submodule update --init --recursive`.  After running that the compilation seems to be working.

Thanks!
👍

sommersoft, 2018-03-29:
The wonders of automation. The GitHub bot posts things into the Discord channel, and I happened to see it. 🤖 

Glad you're rolling now! And thank you for the issue!

tannewt, 2018-03-30:
Someone want to fix the README and fix this issue?

sommersoft, 2018-03-30:
I'll grab it. I was waiting to see if you or Dan were going to put any additional labels or milestones on it.

dhalbert, 2018-03-31:
FIxed by #727. Thanks @sommersoft!

closed 2018-03-31 by dhalbert