adafruit/circuitpython #726: Compiling and including libraries

srfoster, 2018-03-30:
I'm compiling the atmel-samd for Circuit Playground.  That works.  I can deploy the uf2 file to the device.  However, I'm in a bind:

1) When I compile the 3.x version, I don't have some of the audio libraries.  I get an error on `import audioio`.  I think that this might be expected.  I read that 3.x (alpha) doesn't necessarily have the full API yet.

2) When I compile the 2.x version, I don't have the neopixel library.  I get an error on `import neopixel`.  Is this expected?  I don't know.  Maybe I did something wrong.

3) When I download the current 2.x release, I can import both neopixel and audioio.  However, I have my own mpy libraries that I want to freeze into the firmware, so I can't use this option.  I get a memory allocation error when including my mpy libraries from the file system.  This is why I'm trying to compile from scratch in the first place.

Can someone help me figure out how to make a build that includes my own frozen mpy libraries along with the rest of the awesome API that ships with 2.x?

dhalbert, 2018-03-30:
Which neopixel library do you want? The one named `neopixel` is a Python-based library from https://github.com/adafruit/Adafruit_CircuitPython_NeoPixel. It's not built in; you'd get it from the bundle or otherwise have it on `CIRCUITPY`. The one named `neopixel_write` is a low-level library that is implemented in C and is built in.

srfoster, 2018-03-30:
Thanks for responding!  I hadn't realized there were two libraries.

I was hoping to get the `neopixel` library into the build.

Is `neopixel` really not built in?  I just fully wiped my CPX and re-flashed it with the 2.2.4 version of circuit python that I downloaded from the releases page.  And `import neopixel` does work.  Maybe I'm confused about something here though.  

Also, the Adafruit_CircuitPython_NeoPixel repo is linked inside the `frozen/` directory of the 2.x branch of this repo, right?  I assumed that (somewhere in the Make process) these libraries in `frozen/` were automatically included in the build.  Maybe I am missing a flag to tell it to do this, though?

I will see if I can either 1) figure out if I'm missing some obvious Make flag, or 2) manually compile and freeze the `neopixel` library into my custom build. 

I'm open to other suggestions, though!

dhalbert, 2018-03-30:
Ah, ok. It's not "built in" in the sense of frozen code, but it's included in the frozen modules for that particular board. The list of frozen modules to include is in `atmel-samd/boards/circuitplayground_express/mpconfigboard.mk`. If you haven't changed that, `neopixel` should be importable in your build.

How did you add your own extra frozen modules to your build? If it makes sense, point me to your github repo.

srfoster, 2018-03-30:
So, I cloned this repo fresh and things seemed to work better this time.  So I probably was just doing something silly (like maybe I hadn't pulled the submodules before trying to compile).

Anyway, here were the (correct) steps I just went through, for future reference:

```
git clone https://github.com/adafruit/circuitpython
cd circuitpython
git checkout branch 2.x
git submodule update --init --recursive
cd mpy-cross
make
cd ../ports/atmel-samd
make BOARD=circuitplayground_express FROZEN_MPY_DIR=modules
```

Putting py files in the `modules` directory worked.  And it included the `neopixel` library too.

Thanks for the nudges in the right directions!

dhalbert, 2018-03-30:
Great! Glad to help!

closed 2018-03-30 by dhalbert