adafruit/circuitpython #731: Bring back the interactive mode for REPL

deshipu, 2018-03-31:
In cpython, when it's run in interactive mode, when the main script being executed finishes (or errors, or is otherwise interrupted), we are dropped into the REPL with all the variables from the script available, so we can easily inspect them and work with all the functions and classes that were defined in the script. It's a standard Python behavior, very useful for debugging and ad-hoc experimenting. MicroPython does exactly the same thing by default — since there is no operating system to which it could exit, it acts as if it was always started in interactive mode, which makes a lot of sense.

However, CircuitPython does an equivalent of soft reset as soon as we enter the REPL, completely destroying all evidence we could use from the program that just crashed, and making it impossible to have a `main.py` with some boilerplate imports for experimenting. We can't even do `import main` to get access to the variables, as the module cache is empty, and that will re-run the program from the start. Plus, if it contains an infinite loop (which it usually does), we can't even access it after importing, since breaking the loop with ctrl+c will cause the import to fail.

Breakpoints don't solve that issue, because then I need to know where to put the break, and I need to edit the original file and re-run it, possibly never getting to the same state that I wanted to examine again. I would like to have an option to use the interactive mode — it would help greatly when developing drivers and libraries. Even if it was a compile-time option, it would help greatly.

tannewt, 2018-04-02:
Try commenting out this reset_mp(). https://github.com/adafruit/circuitpython/blob/master/main.c#L303

`reset_mp` is what resets the heap and the top level VM state (via mp_init()) I believe.

deshipu, 2018-04-02:
Thanks, I will give it a go.

deshipu, 2018-04-02:
I wonder if this could also be an option for #184 — have a flag that controls whether to reset it or not after running the file.

tannewt, 2018-04-02:
@deshipu Sounds good to me!

deshipu, 2018-04-02:
I will try and see if I can get that as a compile-time flag first.

closed 2021-04-02 by deshipu