adafruit/circuitpython #763: Add `getpass` module.

tannewt, 2018-04-13:
Like `input` but doesn't echo back the characters.

https://docs.python.org/3.6/library/getpass.html

https://forums.adafruit.com/viewtopic.php?f=60&t=134104

paulswirhun, 2018-05-18:
I made a hacky version of this today to change input() to *not* echo the input data. I think there is another day or two of work required to separate this functionality into getpass the "right way" (not that I'm signing up to do that) but I estimate that an initial version is only about 20 lines of code plus a few configuration lines added per port. Here are some notes; I admit I implemented the easiest version which just changes the behavior of input(), rather than really add a getpass module. It took me a while to understand the micropython codebase ... hopefully you Circuitpython developers already know all this.

- input() is implemented in py/modbuiltins.c: mp_builtin_input(). However, it uses readline() defined in lib/mp-readline/readline.c, the bulk of which is complex due to readline_process_char(). That function does some elaborate processing upon every received character. This function is very application specific and tailored for efficiency, as well as maybe the Micropython developers preferences (it has an emacs mode?!) :).

- As part of the implementation of input(), you can override the (#define mp_hal_readline readline) directive per micropython port, although I don't see that any ports actually take advantage of this. This is too broad of an override, since it also affects everything else that uses readline, including the REPL. The Unix port appears to override readline (few or none of the embedded systems do), but I think the unix port actually just enters then exists some "raw" mode using stubs which appear to do nothing. I didn't look in great detail. Basically, the highly flexible readline implementation that is useful for the REPL is not a great choice for input() and definitely not for getpass().

- You don't want all of readline_process_char() sitting in the getpass imlementation (or the "raw bytes" input mode I'm requesting in the Adafruit forum). In fact, the mp_builtin_input() function only checks for CTRL+C and CTRL+D return values from readline(), although the readline code has many more non-alphanumeric modes it can return; it saves history (inputs and passwords shouldn't be saved to history like the REPL is), arrow key support, page up/down, etc.

- Instead, I implemented a parallel function to readline, which I called readascii or readbytes, and called that from mp_builtin_input(). It is essentially the 10-line version of readline. It returns normally provided the user doesn't type NULL, CTRL+C, or CTRL+D characters and doesn't exceed the buffer size. It could be further restricted to human readable ascii characters, although that's not really necessary for my application.

I think this is the right approach, but needs a little more effort to make something patch-worthy. I think right now input() only uses readline() out of convenience, but really it should use something parallel and simpler; getpass should probably be even simpler than input in terms of the heuristics it applies to the user input.

tannewt, 2018-05-18:
Sounds like you are on the right track @paulswirhun. Please make a PR when you have something close and we'll take a look.

Thanks!

closed 2021-08-10 by microDev1