adafruit/circuitpython #766: AnalogOut does not work for Metro M4 Rev B on current master

siddacious, 2018-04-17:
It would appear that something is awry with AnalogOut on current master for the Metro M4 rev. B as well as my SAMD51G19 based board.
```
import analogio
import digitalio
import board
d12 = digitalio.DigitalInOut(board.D12)
d12.direction = digitalio.Direction.OUTPUT
d12.value = True
dac = analogio.AnalogOut(board.A0)
dac.value = 65535
```
The above should result in 3.3v on D12 and A0 however A0 shows 0v

dhalbert, 2018-04-17:
The pin assignments for A0 and 1 have not changed from rev B to rev F, but the new `AudioOut` support might be doing something to interfere with `AnalogOut`.

On a Metro M4 rev F, the above code works and sends 3.3V to A0.

tannewt, 2018-04-17:
Did you solder together the jumper on the back of the Rev B? The internal voltage reference is broken on the rev A of the SAMD51. The Rev F metro has it connected by default.

closed 2018-04-17 by siddacious