adafruit/circuitpython #767: No USB on Trinket M0

deshipu, 2018-04-17:
It seems that the current master (7f7cbe8efa7d61683872fa93629491bce7ef5135) breaks Trinket M0 — after flashing the device doesn't appear as USB at all, and the LED is purple. It is working with 3.0.0-alpha.4, so the breaking change must be somewhere in between. I'm now trying to find which commit breaks it.

jerryneedell, 2018-04-17:
FYI - I reproduced this on a Trinket_M0  -- I also built for my metro_m4_express_revb and it seems to be working OK.

deshipu, 2018-04-17:
I just tested it with a Feather M0 Express, and it works fine, but it doesn't work with the µGame 10 board — so it probably only affects crystalless boards.

jerryneedell, 2018-04-18:
also fails in same way on gemma_m0.

sommersoft, 2018-04-18:
Bringing some "insight" over from discord. Just want to keep the discussion up-to-date...

Stepping through in debug, Trinket is getting stuck in a loop here:
```
(gdb) bt 
 #0  hri_sysctrl_get_PCLKSR_XOSC32KRDY_bit (hw=0x40000800) at asf4/samd21/hri/hri_sysctrl_d21.h:798
#1  _sysctrl_init_sources () at asf4/samd21/hpl/sysctrl/hpl_sysctrl.c:127
#2  _init_chip () at asf4/samd21/hpl/core/hpl_init.c:80
#3  init_mcu () at asf4/samd21/hal/include/hal_init.h:68
#4  port_init () at supervisor/port.c:123
#5  main () at ../../main.c:229
```

jerryneedell, 2018-04-18:
FWIW - reverting PR_745  ( added RTC) seems to bring Trinket_M0 and Gemma_M0 back to life. They Boot and CIRCUITPY is mountable.

Note: with PR 745 removed, on the Gemma_M0 the dotstar flashes - sort of red/purple until REPL is entered - then is goes white as usual.  Trinket m0 does the normal green fade prior to REPL.

tannewt, 2018-04-18:
This is fixed by #768

closed 2018-04-18 by tannewt