adafruit/circuitpython #769: Delete key broken in REPL - CircuitPython 2.4.4

Sembazuru, 2018-04-18:
I discovered an issue with the REPL, originally reported to the Mu repository as [issue #225](https://github.com/mu-editor/mu/issues/225#issuecomment-381986417). It was suggested that I report this issue at MicroPython. I submitted it there as [issue #3728](https://github.com/micropython/micropython/issues/3728#issuecomment-382274745). They were unable to replicate so suggested that I report this issue here.

Here is what I originally said:

> I have Mu configured to start in Adafruit CircuitPython mode, I really haven't used any other modes.
>
> In the code editor the DELETE and BACKSPACE keys function as expected.
>
> I then click the REPL button to open the REPL/Serial pane. I then press any key to "enter the REPL". I execute a command (for example help("modules")). Then I press up arrow to get the previous command and use the left arrow to move the cursor into the previous command. This is where both DELETE and BACKSPACE both function like BACKSPACE.
>
> If the REPL is running internally on the circuit python device, the issue might actually be with the device firmware instead of Mu. Or, maybe the wrong keycode is being sent for the DELETE key? Here is the version string for the installed circuit python on my device:
> Adafruit CircuitPython 2.2.4 on 2018-03-07; Adafruit Gemma M0 with samd21e18

sommersoft, 2018-04-18:
I can duplicate this with Mu 1.0.0.beta.15 on Win10. However, when using the same steps, it doesn't replicate in PuTTY. 

I'll look into Mu's code a little; IIRC CircuitPython assumes VT100 terminal protocol...

Sembazuru, 2018-04-18:
Thanx. As you can see I started with the Mu folks... Yes, I was using Mu 1.0.0.beta.15, I forget which build. I do mention the download package on the Mu issue.

dhalbert, 2018-04-19:
I started to look at this a bit. The problem shows up only in Windows. DELETE does forward delete in both Linux and MacOS. So it may just be some kind of keymapping problem. I am testing in the plain REPL using a terminal program, not Mu.

sommersoft, 2018-04-19:
@dhalbert yeah, that's what I did earlier. PuTTY showed no signs of reverse-delete. 

I looked at Mu code, but my PyQt knowledge is...that was the first time I read it :smile:. I didn't see anything glaring.

dhalbert, 2018-04-19:
I see the problem with Tera Term on Windows. I didn't have time to try other things. The key mapping in Tera Term looked ok, so I'm going to do a little testing to see what character code it's actually receiving.

sommersoft, 2018-04-19:
@dhalbert I had this thought for a test last night after shutting everything down. What happens when you go to a history item, use `delete`, and then hit enter? What is the resulting exception, and does it match what was showing in the Mu REPL?

Might help narrow down if the problem is the [data being sent](https://github.com/mu-editor/mu/blob/master/mu/interface/panes.py#L175), or the [local representation](https://github.com/mu-editor/mu/blob/master/mu/interface/panes.py#L220).

Although, if you're able to grab the bytes being transmitted either way, that'll be more concrete...

dhalbert, 2018-04-20:
Tested this in terminal programs other than Mu REPL: Delete will delete forward if it sends 0x1B 0x4B 0x33 0x7E (ESC [ 3 ~). Backspace sends 0x7F.

Tera Term, out of the box, sends 0x08 for Backspace and 0x7F for Delete, so it only deletes backward. If I change a setting in Tera Term, I can get it to send the combinations in the paragraph above.

I don't think there's anything to change in MicroPython or CircuitPython: they support the first-paragraph combinations properly.

I posted this in the Mu issue as well, and will wait for feedback there.

dhalbert, 2018-05-03:
This was diagnosed as a Mu issue (see above): https://github.com/mu-editor/mu/issues/225 (fix in process). Closing.

closed 2018-05-03 by dhalbert