adafruit/circuitpython #770: PWM "Timers in use:" error is not descriptive

kattni, 2018-04-19:
We need to change the error thrown when PWM doesn't work on a pin from "Timers in use" to "Pin not supported". The current error is not clear. The pins that fail will never support PWM, and the error should be explicit.

tannewt, 2018-04-20:
Related code is here: https://github.com/adafruit/circuitpython/blob/master/ports/atmel-samd/common-hal/pulseio/PWMOut.c#L179

During the pin search an additional piece of state which indicates whether a possible pin was found needs to be tracked. If no potential pins are searched then return ValueError with the message "Pin not supported" instead of the existing RuntimeError.

dhalbert, 2018-05-08:
Looking at this: at  https://github.com/adafruit/circuitpython/blob/master/ports/atmel-samd/common-hal/pulseio/PWMOut.c#L109 there's a check to see if the pin supports any timers at all. If not, it throws ValueError with "Invalid pin".

Then the code checks to see if any of the possible timers for the pin are available. If not it throws "All timers in use". Maybe that could be better phrased as "All timers for this pin are in use".

Am I missing something?

tannewt, 2018-05-09:
The piece missing from this issue is that TCs with waveout 0 are also never supported since we always run TCs in 16bit mode which uses CC[0] for the top value.

rhooper, 2018-05-14:
I'm going to take a look at this.

dhalbert, 2018-05-14:
Fixed via #833.

closed 2018-05-14 by dhalbert

rhooper, 2018-05-14:
To reproduce and test:

```
Adafruit CircuitPython 3.0.0-alpha.6-75-ga0954b9e1-dirty on 2018-05-14; Adafruit Gemma M0 with samd21e18
>>> import microcontroller, pulseio, board
>>>
>>> pwm1 = pulseio.PWMOut(microcontroller.pin.PA14, duty_cycle=1, frequency=500, variable_frequency=True)
>>> pwm2 = pulseio.PWMOut(microcontroller.pin.PA18, duty_cycle=1, frequency=500, variable_frequency=True)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: All timers for this pin are in use
>>> pwm3 = pulseio.PWMOut(board.D13, duty_cycle=1000, frequency=600)
>>>
>>> pwm4 = pulseio.PWMOut(board.D0, duty_cycle=1000, frequency=60)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: All timers in use
```