adafruit/circuitpython #775: spi: would be nice to have an LSBFIRST flag, like arduino

jepler, 2018-04-21:
e.g., in arduino code you can write `SPI.beginTransaction(SPISettings(SPIRate, LSBFIRST, SPI_MODE0));` but there seems to be no equivalent in CP.  A new kw-only parameter to `SPI.configure` could add this capability backward-compatibly.

This flag would be useful for controlling AD9850-family DDCs from CP; right now it looks like the manual bitswapping coded in Python is a main limitation of the update rate.

dhalbert, 2018-04-22:
MicroPython supports MSB/LSB first. Checking, it appears the nRF52840 and ESP8266 peripherals support a choice of data ordering.

jepler, 2018-04-22:
Interesting. I was looking at the bitbang implementation first of all, not hardware implementations. What would the circuit python philosophy be if some HW doesn't support it? Allocate extra buffers for swapped data, it tell the user to use the bitbang version via an exception?

dhalbert, 2018-04-23:
I looked more closely. MicroPython `machine.SPI` only supports MSB first, and will complain otherwise. `pyb.SPI` supports both. It supports 8 or 16-bit writes, presumably due to the chip. Atmel supports 8 or 9 bit writes. I'm inclined to think if we added this, we would restrict the capabilities based on the chip involved rather than fall back to bitbang silently.

tannewt, 2018-04-23:
+1 to throwing an exception and pointing to bitbangio in the docs.

jepler, 2020-08-18:
We did implement this but in the end as there are few devices that need it we decided not to incorporate it.  The code will remain attached to the closed PR https://github.com/adafruit/circuitpython/pull/3224

closed 2020-08-24 by jepler